ZoneMinder/zoneminder #1827: Add an API to authenticate to zms and the API without sending passwords with every API request

mnoorenberghe, 2017-03-20:
While working on the Home Assistant integration I realized that the API hasn't really been designed with security and ease-of-use in mind when it comes to authenticating to the API and authenticating to ZMS streams (when ZM auth is enabled).

Problems with the [current approach](https://zoneminder.readthedocs.io/en/stable/api.html#security):
1. API Consumers are required to forge a login form POST request to start a session. This gets in the way of CSRF protection (#1822) since the consumer would have to scrap the nonce.
2. It's really difficult, if not impossible, to calculate the auth hash for use in the zms URL. Without that, the zms URL needs the plaintext username and password query parameters which show up in server logs.

From the discussion with @pliablepixels on IRC and my experience with the HA integration, I think there should be an API to do the following:
Given a username and password in a POST, return the auth hash for the user.

That alone would fix (1) and (2) above for consumers who only want do get a ZMS stream but wouldn't help with consumers who want to use other API endpoints (assuming it didn't start a session). To protect the API endpoints against CSRF we can also support using the auth hash for API authentication.

I'll submit a patch this week (hopefully later today) to implement this. I still have to think through the various config options related to auth. to hopefully have this approach work for all combinations:
* OPT_USE_AUTH
* AUTH_TYPE
* AUTH_RELAY
* AUTH_HASH_IPS
* AUTH_HASH_LOGINS

(It would be nice to remove some of the options e.g. AUTH_HASH_LOGINS to simplify things).

mnoorenberghe, 2017-03-21:
WIP branch at https://github.com/ZoneMinder/ZoneMinder/compare/master...mnoorenberghe:api_login?expand=1

knnniggett, 2017-03-23:
Thank you for your help.

The two who have used the API the most, and are most familiar with the inner workings, are yourself and @pliablepixels . I'm not sure @kylejohnson has been involved with the API much since he wrote it. Consequently, I think the most sensible thing to do here is let you drive this ship.

I basically feel like a middleman here. I recently learned enough to implement the csrf-magic library, but I haven't used the API enough to have an answer as to how to implement it w/o breaking the API.

In hindsight, I think we all would like the API to not depend on the web frontend at all, but I think the scope of that is probably too big to implement at this time.

Eliminating redundant features sounds good, but of course we need to be careful how we handle that ....will need to make sure a clear migration path is documented.

knnniggett, 2017-03-23:
I was working with @pliablepixels the other day over the possibility of getting zmninja to use AUTH_HASH_LOGINS. In order to do that, the client must the hash in the same way the getAuthUser php function works:
https://github.com/ZoneMinder/ZoneMinder/blob/master/web/includes/functions.php#L129

So to generate the key, the client has to know AUTH_HASH_SECRET, the username, the password, (optional) remote ip address, and several elements of time.

Not saying we have to do it this way, but it seems possible.

mnoorenberghe, 2017-04-02:
> So to generate the key, the client has to know AUTH_HASH_SECRET, the username, the password, (optional) remote ip address, and several elements of time.

That's actually what my patch is implementing :)

mnoorenberghe, 2017-04-03:
I'm not honoring AUTH_HASH_LOGINS in #1845 since that seems to be about the web and will be a support issue if the API relies on that. This approach only works when ZM_AUTH_RELAY == "hashed" since `getAuthUser` checks that but I don't really think it needs to but personally I think everyone should use "hashed" and this is at least a starting point for sane API access.

mnoorenberghe, 2017-04-03:
@pliablepixels Does this work for you? It allows using the usual `auth` query parameter for any API call.
[My initial WIP](https://github.com/ZoneMinder/ZoneMinder/compare/master...mnoorenberghe:api_hash_secret) was also adding an API to calculate the hash for the API consumer so that it doesn't need to know the algorithm itself and then it handle the IP address for you which I know you mentioned is a problem and I agree. I can implement that in a separate pull request if it makes sense but I think just supporting `auth=` is a start without that.

pliablepixels, 2017-04-03:
@mnoorenberghe I liked your original idea of implementing a new API that allows me to login with u:p and returns a token. I'm a little confused by what you are asking here - I looked at the code and if I understand what is needed is zmNinja needs to be provided with auth_hash and auth_hash enabled in ZM. Given that this is two new things a user needs to enable/provision, even if we assume 1.30.2 bundles this in, its possible its turned off, or misconfigured (especially the hash value at zmNinja side), it might lead to a lot of 'my login stopped working' complaints.

I can always fall back to old methods, but I'd prefer a simple switch which is predictable (if v<1.30.2, fake php login,  else use api login). Will you still be doing what you set out to do in https://github.com/ZoneMinder/ZoneMinder/issues/1827 ? That is ideal because it doesn't need any additional client side config.

nitrag, 2017-05-09:
+1

Why not just have an API key(s).

connortechnology, 2017-05-30:
The PR's merged do not address the issue that there is no way to get the auth key via the API.  So we are only half way there.

michaelarnauts, 2017-11-12:
Why not use something like JWT that is an industry standard for this purpose?

connortechnology, 2018-05-18:
I think this has all been addressed now.  We have an api /zm/api/host/getCredentials.json that returns everything you need.

closed 2018-05-18 by connortechnology

pliablepixels, 2018-05-18:
We still don't have an API to login - should we keep this open till its resolved?

commit 33092e40 (2017-05-16): Allow API authentication using the `auth` query parameter containing an auth. hash. (#1845)

* Allow API authentication using the `auth` query parameter containing an auth. hash.

Fixes #1827

The same auth. hash for zms is used here. This allows consumers to use the API without sending the password in the query string and not require forging logins via the login form.

* Move logger.php's global Debug function to Logger::Debug to avoid polluting globals

This avoids a conflict with CakePHP when logger.php gets included indrectly from API code.

* Protect action=login when ZM_ENABLE_CSRF_MAGIC [... cut]