ZoneMinder/zoneminder #1829: API has write permission issues when run using the Dockerfile

mnoorenberghe, 2017-03-21:
> [Tue Mar 21 00:53:07.200935 2017] [:error] [pid 502] [client 172.17.0.1:37484] PHP Warning:  _cake_core_ cache was unable to write 'cake_dev_en-us' to File cache in /usr/local/share/zoneminder/www/api/lib/Cake/Cache/Cache.php on line 328
> [Tue Mar 21 00:53:07.200970 2017] [:error] [pid 502] [client 172.17.0.1:37484] PHP Warning:  /usr/local/share/zoneminder/www/api/app/tmp/cache/persistent/ is not writable in /usr/local/share/zoneminder/www/api/lib/Cake/Cache/Engine/FileEngine.php on line 385
> [Tue Mar 21 00:53:07.201022 2017] [:error] [pid 502] [client 172.17.0.1:37484] PHP Fatal error:  Uncaught exception 'CacheException' with message 'Cache engine "_cake_core_" is not properly configured. Ensure required extensions are installed, and credentials/permissions are correct' in /usr/local/share/zoneminder/www/api/lib/Cake/Cache/Cache.php:186\nStack trace:\n#0 /usr/local/share/zoneminder/www/api/lib/Cake/Cache/Cache.php(151): Cache::_buildEngine('_cake_core_')\n#1 /usr/local/share/zoneminder/www/api/app/Config/core.php(376): Cache::config('_cake_core_', Array)\n#2 /usr/local/share/zoneminder/www/api/lib/Cake/Core/Configure.php(72): include('/usr/local/shar...')\n#3 /usr/local/share/zoneminder/www/api/lib/Cake/bootstrap.php(431): Configure::bootstrap(true)\n#4 /usr/local/share/zoneminder/www/api/app/webroot/index.php(94): include('/usr/local/shar...')\n#5 /usr/local/share/zoneminder/www/api/index.php(41): require('/usr/local/shar...')\n#6 {main}\n  thrown in /usr/local/share/zoneminder/www/api/lib/Cake/Cache/Cache.php on line 186

I think this has something to do with file permissions of /usr/local/share/zoneminder/www/api/app/tmp but I see that directory is already owned by "www-data" and that's what the apache processes are running as.

On a related note, it would be good to have instructions for devs at https://zoneminder.readthedocs.io/en/stable/contributing.html on how to test their changes. I made many fixes to https://github.com/ZoneMinder/ZoneMinder/wiki/Docker but given this issue I guess it's not the common way of developing.

knnniggett, 2017-03-21:
Setting permission on the cake tmp folder is handled by packaging scripts. Changing permissions on files cannot be done with cmake, because that breaks (all) packaging. Consequently, the best solution I can see for this is add an appropriate command in zmlinkcontent.sh to either chmod the cake tmp folder or symlink it somewhere else like /tmp. The former might be better since Docker probably doesn't have a lot of room under /tmp. zmlinkcontent.sh is only used for from-source builds.

What you are likely to find is that each of us likely test pr's slightly differently, which I see as a good thing as that can help uncover problems when using one particular method over another.

I test all pr's by building a package and then install that. Once a build environment is set up, it's not that hard. Instructions for that are in our readthedocs sites for popular redhat & debian distros. I don't ever build from-source anymore. It's too easy for a mistake to wreck your system and clean make uninstalls are no longer possible. Hope that helps.

knnniggett, 2018-01-17:
fixed in commit c0955e00430292ac55907a81059e23056464afbf

Will test tomorrow.

Note the dockerfile has been moved to its own repo: https://github.com/ZoneMinder/zmdockerfiles

closed 2018-01-17 by knnniggett