ZoneMinder/zoneminder #1833: Zone Minder looses tracks of events

doublehp, 2017-03-22:
For camera 14, Zoneminder indicates 2456 total events, and 2456 events for the last month.

I have set a ZM filter to remove all events older than 30 days (without any condition).

I have written a munin plugin to count events stored in the disk. It finds 2.46k events for the last month, and 2.60k for the last year.

But, since you said ZoneMinder is mature, and unlikely to have undiscovered bugs, the issue is probably around ls | wc, unless the filesystem dreams about 140 unexistant folders, or MySQL hiding DB entries to ZM. Please, close this bug ASAP without fixing it as you did for the previous ones. Yes, please, just mark it as DUP of #1748, will save you time and energy.

Clue: if nobody ever found this bug, and there is trace of similar issue in the forum, maybe it could be because nobody ever took time to check the filesystem for undeleted events.

zmaudit.pl -v
1.30.0

cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux stretch/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

SteveGilvarry, 2017-03-22:
I have read #1748 and what you describe seems to be a complex, non-standard install, if I am honest I got lost in what is where doing what. Not saying you are not seeing this problem and that it is not a bug, but unless I invest time in studying and replicating your particular circumstances and environment it is difficult to confirm or replicate. Ideally we could replicate it in a standard install with NFS, are you willing to attempt this? Since you have spent time and effort monitoring events folders on your end I hope you are willing to keep investigating.

I am not aware of anyone else reporting this issue before? You alluded to a trace in the forum, join me in the forum to discuss further, where you can provide me with a link to said trace. Once we are able to confirm the existence of a ZM bug I will be happy to log a GitHub issue, and chase it down. Consider the forum our triage process.

I am closing this call until we can verify it occurs in a less customised installation through forum based discussion. I am sure you mentioned using a symlink on events and images to another drive, please review your setup in reference to this document.
https://wiki.zoneminder.com/Using_a_dedicated_Hard_Drive

I am willing to offer you some of my personal time and energy for free. Bear this in mind before you hit send on sarcasm or frustrated vents.

closed 2017-03-22 by SteveGilvarry

doublehp, 2017-03-23:
Ok, let's turn sarcasm off and be serious.

Installation from ZM point of view is standard: aptitude install zoneminder. Then, hours playing in the webinterface to add cameras.

What is custom is the fact storage is not done on the promary disk, but on a network disk. ZM is running inside a virtualbox, and /dev/sdb is the once from the VB host. Software communication between host and guest is done via virtual networking. NFS failed for various reasons (spent 3d on it, it has bugs in my Debian; in particular, it's missing support for locks); Samba also failed, so, I am left with sshfs; I had clear evidences that NFS and Samba were broken; I never had any clue that could indicate SSHFS as being responsible for any trouble. You can try to replicate with NFS, but you *WILL* have locking issues (I am not sure if it's ZM or an other soft than uses locks, but if ZM uses locks, you are doomed to fail with Debian Stretch - don't try to report them, it's 6 months I am fighting with a service that does not start at all, broken feature in a working service is not an interesting fight).

The symlink is to help ZM find it's storing folder:
# ll /var/cache/ | grep zone
lrwxrwxrwx  1 root root   49 Oct 11 09:24 zoneminder -> /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder
The move was done while ZM service was stopped. The move was done in november, so (see below) lost events from mid february 2017 can not be related to any move problem. If my storage was that bad, I would loose much more than a few events per week. Destination of link is the network mounted folder; physically, it's the same machine ... since ZM is running on a VB guest. Your page says linking is bad; if it was so bad, I would loose much more events than I do; will use the bind soon.

Your tuto has a typo: on my debian, the apache user is not apache but www-data:www-data.

And cameras added after my migration were not configured properly by ZM:
# ll /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/
total 28
drwxr-xr-x 3 www-data www-data 4096 Jan 26 00:11 1
drwxr-xr-x 3 www-data www-data 4096 Jan 30 00:14 11
drwxr-xr-x 2 www-data www-data 4096 Sep  2  2016 13
drwxr-xr-x 3 www-data www-data 4096 Jan 16 20:43 14
drwxr-xr-x 3 root     root     4096 Jan 30 00:14 17
drwxr-xr-x 3 root     root     4096 Jan 16 15:19 18
drwxr-xr-x 4 www-data www-data 4096 Jan  1 00:29 8
lrwxrwxrwx 1 www-data www-data    1 Aug 16  2016 NVT_ZM-08cam_04_jidetech_portail_40 -> 8
lrwxrwxrwx 1 www-data www-data    2 Sep  2  2016 NVT_ZM-14_cam-06-top-201_rue_42_depuis_gite -> 14
lrwxrwxrwx 1 root     root        2 Dec 24 20:00 NVT_ZM-17_cam-07-Sance_cours_38 -> 17
lrwxrwxrwx 1 root     root        2 Jan 16 15:14 NVT_ZM-18_cam-08-Sance_jardin-nord_38 -> 18

=> I don't understand why your page asks to run chown -R apache:apache /newdrive/zoneminder while cameras added after my migration are owned by root (and they work fine in the web interface) (again, if this was critical, I would have hundreds of problematic events per week, not just 1 or 2)

What is completely non standard is that I wrote scripts to monitor everything, including, tracking bugs of all software. The root issue is that, on failure, ZM is not automatically restarted, and internal errors are not reported (they are sent to syslog which is silent). So I developped external monitoring tools (based on Munin) to keep track of all problems (loggrep on log files to scan for frequent errors, detect disconnected cameras, and count files on disk). That's why I am finding bugs no-one ever saw; because I wrote "bug trackers": a single line containing the ERROR word, in any log file in /var will put a red alarm during 1 week. Concerning ZM, this is usefull to detect when process dies and is not restarted, or, when a camera get's broken; for more general use, this is vital to detect SMART failing flags on a HDD and replace them before loosing data (the only data I lost in 19 years were on hardware that don't report SMART).

Why is this important ? ZM cleaning tool is leaving files on disk; these files are not deleted by internal tools; not many, but some. With time, those left over files will use more and more room on disk, and the "free space" usable for live events will reduce month after month. After a few months or years, usable space may become half, or quarter of real disk size; reducing the history of viewable events, or, keeping images for a longer time than the local laws allows it (French law states images must be deleted after one month; I have set a filter at 30 days, and, Munin says I have 140 events older than 1 month. These images are out of database, so, they can not be accessed via ZM interface; they are visible when you use a file browser). So, I could fix it with a radical counter measure, and, use a find -cdate +30 thing to clean your mess; or, report here to make you fix the problem for the whole community. The problem is known on forums, and did not led to any fix. 

I see this this morning in my cron logs:
Removing old day (45) containing 1 events: '/mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/18/17/02/06'
disabling my cleaning script to keep those events a bit longer. This output means: day D-45 contained one lost event.

Out of topic: got this several times a day:
Message from syslogd@leon-03 at Mar 23 05:48:34 ...
 n't kill a non-numeric process ID at /usr/bin/zmdc.pl line 558, <CLIENT> line 1.
make
and, dispite what I was said previously, I still think there is a bug in ZM causing this. A bug most people can't see either because they are using Windows which doesn't report anything, or, are not permanently loggin the server use for ZM.

kingofkya, 2017-03-23:
Chimes in:
As some one who has worked at a hosting comanny supporting about ever ridiculous os/software combo. 

Honestly NFS is prime suspect number 1 if there is any weirdness on the network packet loss overloaded  links/cpu etc.. (non zm experience)  your going to have a bad day.

With NFS it can be tricky to get right in any high load environment, Its really common of files to be modified / removed and the operation dosen't succeed for one reason or another, and Zm likely can't wait for that nonsense.

your adding a ton of extra latency at a min, and that not good at all with the way zm uses its data store.  

And honestly if you really want to narrow the issue down priority one is remove virtual box from the equation. (There are way too manny driver issues with it in production environment)  Make a disk image and put it on real hardware and see if the issue persists.  Or at lest in another environment xen, KVM etc.

Advice/questions you need to answer for yourself.
Does this vm crash or lock up occasionally? (are io operations not being finished )
Is nfs set to "sync" insead of async? (async will store data in a buffer and wait to write)
Have you done a net/wireshark capture to see what nfs is doing?
Is the Host Server/pc or NFS ever maxed or nearly maxed out? 
Hard disks are cheap see if your issue continues with local storage.

doublehp, 2017-03-24:
Note that my NFS (SSHFS for now) is not going throught any copper line; only used between a VirtualBox guest and it's host. Never heard so hard criticism about NFS, it's usually said to be the best of it's kind. But, I have found bugs in VirtualBox when you pass a raw partition to a guest; next time, I will try a raw disk.

I will not try ZM out of VB; no time to setup your test before months (I am 2 years late on more important things).

Not using NFS at the moment, but SSHFS (which is said to be worse, but, Munin and syslog never reported any issue about it, and, I have wrote scripts to track issues in there). I had tried sync and async; the issue is that in Debian stretch, NFS locks don't work.

As said in previous bug reports, yes the VM does crash, once a day. But I have not yet established a correlation between the lost events, and the system crashes (did try to check if the lost event matches a crash at the time of even creation (could do that, did not do it yet), or at the time of event deletion (no clue how to check that: I would need to keep a log of when ZM is removing events from the DB, if event had been put in there)). Since the issue reveals itself only 15 or 45 days after the cause, its complicated.

Wireshark: no.

Maxed out: no. Not anymore. Was an issue until december, I fixed that.

Whatever you may think, if the bug is happening at even deletion time, I think that ZM first removes the entry from the DB, and then, deletes the files from disk; if a crash occurs between the two, it's normal that zmaudit.pl misses the glitch. Check the code for this detail. If I am right, the fix is trivial: remove file from disk, BEFORE removing entry from DB; if a crash occurs between the two, zmaudit.pl will detect missing file and remove event.

If glitch occurs at even creation time, you need to make zmaudit.pl counter check that every image/event folder belonws to a DB event. Again, it depends in which order things are done; put things in an order such that, if a crash happens, zmaudit.pl can clean the mess later.

I am 100% certain there is an easy fix for this bug.

doublehp, 2017-03-24:
Here is an output of my scanner: it states how old a day is (values above 30 means, ZM filter failed to remove files), and how many events are found for that day (was a security to track a special case, not relevant here).

old day (42) containing 1 events: '/mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/17/17/02/10'
old day (41) containing 1 events: '/mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/14/17/02/11'
old day (42) containing 1 events: '/mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/14/17/02/10'

And crash listing for the corresponding days: This is a list of boot times; could designate either a system crash, or a classic reboot. It's impossible to make the difference with the following trivial filters:

(I will embold mont-day-hour-minute, not sure this website will show bold words)

# zcat syslog-20170210.gz syslog-20170211.gz syslog-20170212.gz
2017-**02-10T05:03**:47.472941+01:00 leon-03 kernel: [    0.000000] Linux version 4.8.0-2-amd64 (debian-kernel@lists.debian.org) (gcc version 5.4.1 20161019 (Debian 5.4.1-3) ) #1 SMP Debian 4.8.11-1 (2016-12-02)
2017-**02-10T06:13**:42.437172+01:00 leon-03 kernel: [    0.000000] Linux version 4.8.0-2-amd64 (debian-kernel@lists.debian.org) (gcc version 5.4.1 20161019 (Debian 5.4.1-3) ) #1 SMP Debian 4.8.11-1 (2016-12-02)
2017-**02-11T12:01**:13.546482+01:00 leon-03 kernel: [    0.000000] Linux version 4.8.0-2-amd64 (debian-kernel@lists.debian.org) (gcc version 5.4.1 20161019 (Debian 5.4.1-3) ) #1 SMP Debian 4.8.11-1 (2016-12-02)

An other method, which provide identical results: stating line:
# zcat syslog-20170210.gz syslog-20170211.gz syslog-20170212.gz  | grep -a -e "rsyslogd" -B50 | grep -e "] start"
2017-**02-10T05:03**:47.470876+01:00 leon-03 liblogging-stdlog:  [origin software="rsyslogd" swVersion="8.23.0" x-pid="456" x-info="http://www.rsyslog.com"] start
2017-**02-10T06:13**:42.436255+01:00 leon-03 liblogging-stdlog:  [origin software="rsyslogd" swVersion="8.23.0" x-pid="454" x-info="http://www.rsyslog.com"] start
2017-**02-11T12:01**:13.544711+01:00 leon-03 liblogging-stdlog:  [origin software="rsyslogd" swVersion="8.23.0" x-pid="455" x-info="http://www.rsyslog.com"] start

The crash detector waits 20mn before rebooting the system; we have crashes, we have orphan files; have orphan files been recorded about 5 to 20mn before a crash-reboot ?

# for i in  /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/14/17/02/10 /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/17/17/02/10 /mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/14/17/02/11 ; do find ${i}/*/*/* -maxdepth 0 ; done
/mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/14/17/**02/10/06/19**/47
/mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/17/17/**02/10/06/2**0/47
/mnt/Leon_Host_Big2/leon-03/_var_cache_zoneminder/events/14/17/**02/11/12/50**/52

No. Orphans happen long after a reboot, not before a crash. We have an orphan on feb 11th at 12:50, and system have been fine the whole rest of the day (syslog file provides lines all day long, untill next logrotate in the 12th at 6AM ... without any crash, shutdown, or reboot)

The bug is not located around event creation.

It's definitely around event removal by filters. I can not know when these 3 events should have been removed; and I have no clue how to generate a log that would produce such a peace of information.

connortechnology, 2017-03-24:
That is exactly what zmaudit does.  It counts every single event on disk, and every single event in the db, and compares, and deletes anything that should be deleted.  If a crash happened during a zmaudit run, then the next zmaudit run would clean it up.

As to filters, ZM can be configured to only delete from the DB, and let zmaudit do the cleanup later.  Perhaps you have this turned on? It is called OPT_FAST_DELETE and is on the System tab of the settings. Still, zmaudit will delete anything left behind.

Run zmaudit and look at it's log file.  Turn on debugging in the logging section of the options.  Are there files in these dirs?  I have found that ZM can leave empty directories around.  My branch adds an extra step to delete them, but that is not in master.

doublehp, 2017-03-26:
In #1748 I said that "Fast unticked".

"That is exactly what zmaudit does." obviously, it does not always do it properly. If it had, I would not be reporting it does not ... (don't forget cron runs it every 10 mn, and, I wait 5 days before higlinting errors; that gave him about ... 6*24*5=720 executions before rising a red flag).

I have said in bug #1748 I do not understand where zmaudit puts logs. Repasting the last mentionned debug log:
```
# cat /var/log/zm/zm_debug.log.29595 | pastebinit
http://slexy.org/view/s2a7GGmMtD
```

SteveGilvarry , your tuto https://wiki.zoneminder.com/Using_a_dedicated_Hard_Drive broke my machine due to a glitch between ZM, sshfs and mountall: mountall started by mounting all local filesystems, so, it did the bind before SSHFS, then, sshfs failed due to non empty folder, and ZM started writing events directly in /mnt (the physical local disk), and filled it up. If /usr/bin/zmaudit.pl works correctly, it should have removed all events from the SQL DB dues to missing files; and when I will restore the mount point, it should remove all files because they had been removed from the SQL DB. My old symlinks was just working fine ... Now, let say i don't care loosing a few events, and try to completely follow your peaces of advice: how do I fix the glitch between mountall and fuse ? mountall tries to mount bind first because it's local, and then, sshfs fails due to non empty dir. In short, how do you use ZM with a NAS in Debian the clean way ? you said me to not use symlinks, and, all other ideas I have will seem even worse (mounting manually everything in rc.local, and then start ZM, keeping the service DISABLED from systemctl point of view). Or, write a couple of .service files and mess the dependency tree to make ZM depend on mount-bind.service, and mount-bin.service dep on mount-network ...

doublehp, 2017-03-27:
I just saw that in the day folder, there are hidden symlinks from strange numbers that could look like event #, to hour-minute-second subfolders. If zmaudit makes a cross check between the SQL DB and the fiels on disk, which part of the disk does it look at ? the symlink, or the folder ? My issue could be that it just removes the link, and sometimes fails to remove the directory, and then, it can not relate the physical subdir to any event. I don't think the bug could happen in the opposite way. Does zmaudit tries to remove first the dir or the link ?

connortechnology, 2017-03-27:
The symlinks are indeed the event #'s.  
The sequence of events when deleting an event are: 
#1 read the link to find out where the files are
#2 rm -r that dir
#3 delete any empty directories above that.

It doesn't seem to actually delete the symlink, but a subsequent pass of zmaudit should detect the dangling link and remove it.

connortechnology, 2017-03-27:
Under Options -> Logging clear the ZM_LOG_DEBUG field.  I find that to be totally annoying. Once that is cleared, the debug logs will go in /var/log/zm/zmaudit.log

doublehp, 2017-03-28:
I have a lead.

For all so called "lost events", the folder is here, but not the link in day folder.

And I have found a repro.

As mentionned, when I tried to convert my storage system from the sym-link approach to the bind recomended above, I met a few bugs (Debian bugs in systemd-mount init script - off topic here). In the end, my server spent 2 days with ZM writing events on / instead of NFS (actually SSHFS), and filled up my disk in hours. What's interesting here is that, during those two days, ZM worked fine for abut 4h (before filling /). During that time, zmaudit had lost tracks of the real storage place (the one containing 500GB for the last month), and thus, removed all events from the SQL database. When I fixed my storage issues, I took care to copy all new files on the large disk (with zm and zmaudit stopped - or, tried to). When old large storage was reconnected correctly, zmaudit had files on disk, without events in SQL. And this is where it's becoming a repro: here is the actual state of my disk:

```
# ls -lha /var/cache/zoneminder/events/14/17/03/*
/var/cache/zoneminder/events/14/17/03/03:
total 28K
drwxr-xr-x 1 www-data www-data  20K Mar 13 17:13 .
drwxr-xr-x 1 www-data www-data 4.0K Mar 28 00:01 ..
drwxr-xr-x 1 www-data www-data 4.0K Mar 12 20:53 08

/var/cache/zoneminder/events/14/17/03/07:
total 20K
drwxr-xr-x 1 www-data www-data  12K Mar 15 17:04 .
drwxr-xr-x 1 www-data www-data 4.0K Mar 28 00:01 ..
drwxr-xr-x 1 www-data www-data 4.0K Mar 15 14:23 07

/var/cache/zoneminder/events/14/17/03/09:
total 28K
drwxr-xr-x 1 www-data www-data  20K Mar 16 19:24 .
drwxr-xr-x 1 www-data www-data 4.0K Mar 28 00:01 ..
drwxr-xr-x 1 www-data www-data 4.0K Mar 16 16:53 10

/var/cache/zoneminder/events/14/17/03/26:
total 12K
drwxr-xr-x 1 www-data www-data 4.0K Mar 27 22:40 .
drwxr-xr-x 1 www-data www-data 4.0K Mar 28 00:01 ..
drwxr-xr-x 1 www-data www-data 4.0K Mar 27 19:20 01

/var/cache/zoneminder/events/14/17/03/27:
total 2.4M
drwxr-xr-x 1 root     root      16K Mar 27 23:59 .
drwxr-xr-x 1 www-data www-data 4.0K Mar 28 00:01 ..
drwxr-xr-x 1 root     root     4.0K Mar 27 04:57 04
drwxr-xr-x 1 root     root     4.0K Mar 27 05:58 05
drwxr-xr-x 1 root     root     4.0K Mar 27 06:59 06
drwxr-xr-x 1 root     root     4.0K Mar 27 07:59 07
drwxr-xr-x 1 root     root     4.0K Mar 27 08:59 08
drwxr-xr-x 1 root     root     4.0K Mar 27 09:59 09
drwxr-xr-x 1 root     root     4.0K Mar 27 10:58 10
drwxr-xr-x 1 root     root     4.0K Mar 27 11:54 11
drwxr-xr-x 1 root     root     4.0K Mar 27 12:59 12
drwxr-xr-x 1 root     root     4.0K Mar 27 13:58 13
drwxr-xr-x 1 root     root     4.0K Mar 27 14:59 14
drwxr-xr-x 1 root     root     4.0K Mar 27 15:59 15
drwxr-xr-x 1 root     root     4.0K Mar 27 16:59 16
drwxr-xr-x 1 root     root     4.0K Mar 27 17:28 17
lrwxrwxrwx 1 root     root        8 Mar 27 04:48 .170838 -> 04/47/52
lrwxrwxrwx 1 root     root        8 Mar 27 04:52 .170839 -> 04/51/41
lrwxrwxrwx 1 root     root        8 Mar 27 04:55 .170841 -> 04/55/21
lrwxrwxrwx 1 root     root        8 Mar 27 04:57 .170842 -> 04/57/32
lrwxrwxrwx 1 root     root        8 Mar 27 05:18 .170847 -> 05/17/41
lrwxrwxrwx 1 root     root        8 Mar 27 05:20 .170848 -> 05/19/43
[...]
```

For march 03, 07, 09 and 26 ... my server probably had crashes, but as you can see, I already had orphan events for march 26th. The rest of the day had been cleaned due to missing entries in SQL, but there are still one or two subfolders remaining for march 26th around 1AM ... this may be helpfull for two reasons: I now know what to look for immediately after a system crash, and may be able to determine immediately which recent event may become orphan, without having to wait 45 days.

So, I am looking for an event subfolder, that is not pointed by any symlink from the parent day folder. And if I can confirm this, we may hold a fix you can merge with trunk. No time now, see you in a few days.

doublehp, 2017-03-28:
Event folder usually also have a hidden file, with the # of event. Orphan folders may have it or not. In most cases, they lost it; in some cases, they still have it. In either case, I am speaking about old events (feb 7th or 8th) - I am aware zmaudit refuses to deal with "fresh" events.

Here is a quick script everyone can run to check if any of you has some orphans around. It counts, per day, the number of links, and subfolders: 

```
# for day in /var/cache/zoneminder/events/[0-9]*/*/*/* ; do nd="$(ls -lha "$day"/*/*/ | grep "^d" | grep -v -e "\.$" -e "^total" -e "^/var" | wc -l)" ; nl="$(ls -lha "$day" | grep "^l" | wc -l)" ; echo -n "$i : $nd - $nl : " ; [ $nd -eq $nl ] && echo "fine." || echo "CHECK IT" ; done
/var/cache/zoneminder/events/11/17/03/27 : 12 - 12 : fine.
/var/cache/zoneminder/events/11/17/03/28 : 12 - 12 : fine.
/var/cache/zoneminder/events/1/17/03/27 : 4 - 4 : fine.
/var/cache/zoneminder/events/1/17/03/28 : 3 - 3 : fine.
/var/cache/zoneminder/events/14/17/02/07 : 137 - 0 : CHECK IT
/var/cache/zoneminder/events/14/17/02/10 : 1 - 0 : CHECK IT
/var/cache/zoneminder/events/14/17/02/11 : 1 - 0 : CHECK IT
/var/cache/zoneminder/events/14/17/02/16 : 5 - 0 : CHECK IT
/var/cache/zoneminder/events/14/17/03/03 : 1 - 0 : CHECK IT
/var/cache/zoneminder/events/14/17/03/07 : 1 - 0 : CHECK IT
/var/cache/zoneminder/events/14/17/03/09 : 1 - 0 : CHECK IT
/var/cache/zoneminder/events/14/17/03/26 : 1 - 0 : CHECK IT
/var/cache/zoneminder/events/14/17/03/27 : 582 - 582 : fine.
/var/cache/zoneminder/events/14/17/03/28 : 190 - 190 : fine.
/var/cache/zoneminder/events/17/17/02/10 : 1 - 0 : CHECK IT
/var/cache/zoneminder/events/17/17/02/27 : 1 - 0 : CHECK IT
/var/cache/zoneminder/events/17/17/03/13 : 1 - 0 : CHECK IT
/var/cache/zoneminder/events/17/17/03/27 : 88 - 88 : fine.
/var/cache/zoneminder/events/17/17/03/28 : 17 - 17 : fine.
/var/cache/zoneminder/events/18/17/02/16 : 3 - 0 : CHECK IT
/var/cache/zoneminder/events/18/17/03/27 : 44 - 44 : fine.
/var/cache/zoneminder/events/18/17/03/28 : 25 - 25 : fine.
/var/cache/zoneminder/events/8/17/02/16 : 1 - 0 : CHECK IT
/var/cache/zoneminder/events/8/17/03/07 : 1 - 0 : CHECK IT
/var/cache/zoneminder/events/8/17/03/13 : 1 - 0 : CHECK IT
/var/cache/zoneminder/events/8/17/03/27 : 26 - 26 : fine.
/var/cache/zoneminder/events/8/17/03/28 : 13 - 13 : fine.
```

Note sure what to test further: all orphans have lost their parent link, whether they still have their local hidden ref or not.

So the operation that ocasionally fails on my computer is ... according to steps explained by connortechnology, the "rm -r dir", and zmaudit immediately jumps to removing the link.

The easy fix could be: if a day folder does not contain any symlink, remove the day; could this have bad side effects ? like loosing archived events, user labels or other meta data ? If yes, could we move those folder into quarantine somewhere else, in some kind of trash folder (like lost-and-found) ? Add an option in the general conf to let admin decide what to do ?

Also, dispite the fact I have been told 100 times to be carefull to always make folders readable by Apache (www-data), I see that all files are owned by root (see logs above). Does not seem harmfull to me, but could explain why I have to run zmaudit from cron, and auto-removal does not work the normal way ? on this one, it could related to Debian packaging. Should I Report this to them ?

doublehp, 2017-03-28:
At the moment (as it is since 2 months), "Under Options -> Logging clear the ZM_LOG_DEBUG field." is UNticked. 

```
# ls -lha /var/log/zm/zmaudit.log
-rw-r--r-- 1 www-data www-data 4.6K Mar 27 05:23 /var/log/zm/zmaudit.log

# cat /var/log/zm/zmaudit.log
27/03/17 05:05:32.247863 zmaudit[8923].ERR [Unable to unlink '17/17/03/22/.167331': No such file or directory]
27/03/17 05:06:24.333135 zmaudit[14367].ERR [Unable to unlink '17/17/02/28/.148215': No such file or directory]
27/03/17 05:08:54.718420 zmaudit[14367].ERR [Unable to unlink '17/17/03/05/.152732': No such file or directory]
[...]
27/03/17 05:23:36.101806 zmaudit[10755].ERR [Unable to unlink '8/17/03/16/.161817': No such file or directory]
27/03/17 05:23:36.121862 zmaudit[21549].ERR [Unable to unlink '8/17/03/16/.161817': No such file or directory]
27/03/17 05:23:36.144546 zmaudit[4257].ERR [Unable to unlink '8/17/03/16/.161817': No such file or directory]
```

From this error log, one line matched a day from my scanner:
```
27/03/17 05:23:13.955300 zmaudit[3221].ERR [Unable to unlink '8/17/03/13/.159219': No such file or directory]

# ls -lha /var/cache/zoneminder/events/8/17/03/13/
total 12K
drwxr-xr-x 1 www-data www-data 4.0K Mar 27 05:23 .
drwxr-xr-x 1 www-data www-data 4.0K Mar 28 00:07 ..
drwxr-xr-x 1 www-data www-data 4.0K Mar 27 05:23 15

# ls -lha /var/cache/zoneminder/events/8/17/03/13/15/20/16/
total 13M
drwxr-xr-x 1 www-data www-data  12K Mar 13 15:21 .
drwxr-xr-x 1 www-data www-data 4.0K Mar 13 15:20 ..
-rw-r--r-- 1 www-data www-data 178K Mar 13 15:21 00170-capture.jpg
-rw-r--r-- 1 www-data www-data 179K Mar 13 15:21 00171-capture.jpg
-rw-r--r-- 1 www-data www-data 181K Mar 13 15:21 00172-capture.jpg
[...]
-rw-r--r-- 1 www-data www-data 178K Mar 13 15:21 00237-capture.jpg
-rw-r--r-- 1 www-data www-data 180K Mar 13 15:21 00238-capture.jpg
-rw-r--r-- 1 www-data www-data 182K Mar 13 15:21 00239-capture.jpg
```

So the day is lacking symlink, and the even inside also lacks the hidden reference. It's impossible to tell if this is event is (was) 159219 or not. All other lines from /var/log/zm/zmaudit.log refer to days that have been completely removed.

Now that I have logs, I do find interesting errors in there, in older log files:

```
23/03/17 09:00:24.635094 zmaudit[26901].FAT [Can't open directory '1/17/03/23': Transport endpoint is not connected]
24/03/17 09:52:53.213294 zmaudit[3858].FAT [Can't open directory '11/17/03/04': Transport endpoint is not connected]
25/03/17 05:12:24.613353 zmaudit[1733].FAT [Can't execute log entry 'insert into Logs ( TimeKey, Component, Pid, Level, Code, Message, File, Line ) values ( ?, ?, ?, ?, ?, ?, ?, NULL )': The table 'Logs' is full]
```
This must be when I had disk full 2d ago.

"Transport endpoint is not connected" is when SSHFS fails (I detect and fix it via crontab every 10mn).

I found one match between /var/log/zm/zmaudit.log-20170312.gz and my above log:
```
10/03/17 18:50:34.560430 zmaudit[25937].FAT [Can't open directory '14/17/03/09': Transport endpoint is not connected]
/var/cache/zoneminder/events/14/17/03/09 : 1 - 0 : CHECK IT
```
Impossible to tell if it's related.

See an other log:
```
# zcat /var/log/zm/zmaudit.log-20170306.gz
28/02/17 10:20:40.457992 zmaudit[17701].FAT [Can't open directory '8/17/02/11': Transport endpoint is not connected]
01/03/17 05:30:33.793016 zmaudit[31904].FAT [Can't open directory '8/17/02/13': Transport endpoint is not connected]
02/03/17 21:21:57.655328 zmaudit[26782].FAT [Can't open directory '14/17/02/16': Transport endpoint is not connected]
05/03/17 22:00:45.203129 zmaudit[17507].FAT [Can't open directory '18/17/02/17': Transport endpoint is not connected]
```
None of these days match anything.

Don't know what to do next. I gave you my best guesses in the end of https://github.com/ZoneMinder/ZoneMinder/issues/1833#issuecomment-289680308 and begin of https://github.com/ZoneMinder/ZoneMinder/issues/1833#issuecomment-289691989 . I can not find any correlation between orphans and my crashes, and you could just remove days that no longer have any symlink.

connortechnology, 2017-03-28:
Something to try: edit zmaudit.pl and at the top change MAX_AGED_DIRS to something large like 1000000.  

This is something that I removed from my version, because it is stupid.  It limits how many links get checked.  The default is 10, so only 10 links would get checked/deleted etc.  

If the link doesn't exist, then no, zmaudit will not delete the event.  I will see about adding that.

doublehp, 2017-03-30:
MAX_AGED_DIRS => 1000000; done; don't understand what it's for; wait and see.

Removing orphan events is one aspect of the problem (you can fix it); understanding why they become orphan would be better (but may be related to my crashes).

doublehp, 2017-03-31:
Connor, <SACARSM=ON>you are SuperMan<SACARSM=OFF>. You asked me a question: "There should be a corresponding log line [...]  Please read and respond to my post above.", I have spent half an hour to find a way to send you an answer, and a full hour to write the answer. You took less than 10mn to provide me a small code, that I pasted in my server; took me 20mn to understand what your code does, and, I did not understand it all: no clue where I am expected to find the additionnal output, nevertheless, I spent 20mn to check every single aspect of your suggested modification, and make sure your code would run nicely, and that I would know where to read the output (syslog ? messages ? zmlog files ? SQL Log table ?), and my conclusion was: your code will not produce any output because an other issue is making ZM run on 3 legs; I have been confident enough to provide you ultra private links that have never been published since 2006. And here is your super power: took you exactly 85s to read 98 lines and state that I am ignoring you ? 

# stat /usr/bin/zmdc.pl
  File: /usr/bin/zmdc.pl
  Size: 23950           Blocks: 48         IO Block: 4096   regular file
Device: 813h/2067d      Inode: 1376        Links: 1
Access: (0755/-rwxr-xr-x)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2017-03-31 19:36:22.576337000 +0200
Modify: 2017-03-31 19:35:24.664337000 +0200
Change: 2017-03-31 19:35:24.680337000 +0200
 
And, concerning MAX_AGED_DIRS: 
# stat /usr/bin/zmaudit.pl
  File: /usr/bin/zmaudit.pl
  Size: 23257           Blocks: 48         IO Block: 4096   regular file
Device: 813h/2067d      Inode: 18563       Links: 1
Access: (0755/-rwxr-xr-x)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2017-03-31 10:50:01.166004570 +0200
Modify: 2017-03-30 10:43:17.083845000 +0200
Change: 2017-03-30 10:43:17.095845000 +0200
 Birth: -
# cat /usr/bin/zmaudit.pl | grep 1000000
use constant MAX_AGED_DIRS => 1000000; # Number of event dirs to check age on

Not trusting this log ? You almost have root access on my machine !!!

You are free to refuse to help; but, I am not accepting to be said to ignore people when I almost blindly paste their code without understanding what it does. I am barely able to check your code is not going to erase my system, but my understanding of perl does not go any further.

And, do not reproach me to try to provide you the answer to the question you asked me, when the communication channel was closed by a third person.

Your thoughts don't matter anymore when your code is not called at all. Your thoughts will matter when ZM will accept to run normaly. Still, your code is ready to run for when you take the liberty to restart my ZM !!!

Stating I ignored you several times, when your code and conf values are already in my machine, and, reading 98 lines in 85s ... there is really something wrong with you. I have tried hard to stay calm all day. I am one hair away from aptitude remove zoneminder, and give Amazon 350e for a hardware system.

doublehp, 2017-05-22:
I confirm that all orphan events:
- contain picture files in folder
- miss sym-link pointing to event number

Isaac Connor, in 2 months, the patch you sent me by email never gave any output, and did not have any impact on the issue ("n't kill a non-numeric process").

Each time I get the "n't kill a non-numeric process ID" message, I have a service crash (except at 6AM for cron-daily: i get the message without service crash).

There are lines like
```
17/05/17 00:19:21.349027 zmdc[22822].WAR ['zmc -m 17' has not stopped at 17/05/17 00:19:21. Sending KILL to pid 22922]
17/05/17 00:31:02.800572 zmdc[22822].WAR ['zmc -m 17' has not stopped at 17/05/17 00:31:02. Sending KILL to pid 28549]
```
unrelated to service crashes.

Crashes are always related to lines like
```
17/05/17 12:48:11.509555 zmdc[7805].ERR ['zma -m 1' exited abnormally, exit status 255]
17/05/17 12:48:12.017415 zmdc[7805].ERR ['zmc -m 8' exited abnormally, exit status 255]
17/05/17 12:48:12.549324 zmdc[7805].ERR ['zma -m 8' exited abnormally, exit status 255]
17/05/17 12:48:13.035037 zmdc[7805].ERR ['zmc -m 11' exited abnormally, exit status 255]
17/05/17 12:48:14.069171 zmdc[7805].ERR ['zmc -m 14' exited abnormally, exit status 255]
```

Despite tons of logs archived, I can not relate service crashes with the loss of symlinks from events to event number because logs do not tell when the service is trying to remove events. So, I can not tell if events become orphan when service crashes while trying to remove events. I could, if you tell me where such a log is available.

The machine where ZM is running is now stable: since 3 months, I get uptimes around 4 or 6: system reboots only once a week, but ZM keeps on crashing 4 to 6 times a day. Crash frequency of system or ZM is not related to system load, or the number of cameras. I have added more cameras, and tuned buffers, reached high load, reduced load ... stability of server and ZM are not related to any of these factors.

What seemed to affect stability of things was to change the source mode (some forum said to try Remote if FFmpeg does not produce a good image, and RTP/RTSP if RTP/Unicast fails to produce a clean image). Uptime of server seems affected by these. This leads us to video stream decoding issues ... and ZM being sensible to ffmpeg crashes.