ZoneMinder/zoneminder #1834: zmrepo centos7 armhfp packages

mrflow, 2017-03-22:
There are packages for armhfp for fedora 24/25, but not for armhfp for centos 7[1]. Would it be possible to provide such packages also for centos 7, which is actually also available for a few arm devices[2]?

What and how can I contribute to make this possible?

1: https://zmrepo.zoneminder.com/
2: http://mirror.centos.org/altarch/7/isos/armhfp/

mastertheknife, 2017-03-22:
This is interesting.
I personally have an ARMv7 (Scaleway C1 instance, sadly it doesn't have Neon) server that runs CentOS7 for ARM.
Fedora 24 and 25 are bleeding edge, i think ZoneMinder is better suited for stable, long-term linux distribution such as CentOS.
Does it have to be armhfp? What about other ARMv7 variants? For example on my CentOS7 server:
```
Linux kfir1 4.5.7-std-4 #1 SMP Tue Jul 12 11:00:06 UTC 2016 armv7l armv7l armv7l GNU/Linux
```

A major problem is that EPEL isn't available for CentOS7 on ARM, i think some of our dependencies are from EPEL.

Peeking at the 2nd link, i see there is also AArch64 version. This is more interesting, because its 64bit and doesn't seem to be limited to specific ARMv8 variants.

mastertheknife, 2017-03-22:
Okay, i looked here:
https://dl.fedoraproject.org/pub/epel/7/
It seems there is EPEL for CentOS7, but for AArch64 only.

mrflow, 2017-03-22:
Actually, there is an inofficial, pretty well-hidden epel-repo for armhfp:

https://wiki.centos.org/SpecialInterestGroup/AltArch/Arm32#head-f2a772703b3caa90cc284e01bc87423ce9a87bcd

mastertheknife, 2017-03-23:
> Actually, there is an inofficial, pretty well-hidden epel-repo for armhfp:
> 
> https://wiki.centos.org/SpecialInterestGroup/AltArch/Arm32#head-f2a772703b3caa90cc284e01bc87423ce9a87bcd

Thats cool! I added that repo and it works well on my CentOS7 (armhfp, not sure why uname -a shows armv7l)
So its pretty much definitely possible now.

knnniggett, 2017-03-23:
The zoneminder packages in zmrepo depend on two repos to function: RPMFusion and EPEL. If either one of those repos does not support a particular architecture then neither will zmrepo. 

I took a look at the alt-arch EPEL mentioned in this thread, and unfortunately it does not appear to be a perfect mirror of EPEL. For example, the first two packages I went looking for, perl-sys-mmap and perl-soap-wsdl, were not there. So it would appear EPEL and alt-arch EPEL are similar in name only.

I'm not denying that it would be neat to get zoneminder running on centos on arm hardware. However, the scope of work to do this would be significant. It's much to great to even consider.

I'm closing this issue out because it does not meet the criteria of a bug report. If you feel you have the spare time to start building a lot of packages, then feel free to contact me in the user forum or our irc channel. 

If the architecture support for EPEL & RPMFusion changes in the future, you can be sure zmrepo will include those.

closed 2017-03-23 by knnniggett