ZoneMinder/zoneminder #1840: Filter will not delete Events with NULL endTime

connortechnology, 2017-03-29:
At line 298 of scripts/ZoneMinder/lib/ZoneMinder/Filter.pm we have the following code:
`      if ( $self->{AutoMessage} ) {
# Include all events, including events that are still ongoing
# and have no EndTime yet
        $sql .= " and ( ".$self->{Sql}." )";
      } else {
# Only include closed events (events with valid EndTime)
        $sql .= " where not isnull(E.EndTime) and ( ".$self->{Sql}." )";
      }
`
As you can see, for anything other than AutoMessage we add an additional not isnull(E.EndTime).  The idea is to prevent the filter from acting on events that are still being recorded.

The problem is that Events  can get in this state when zma crashes.  This leads to confusing behaviour where events are not deleted and storage fills up.

I think we shouldn't be adding the extra magic term.  instead we should be adding an explicit term to the filters saying event not finished or something like that.  

Thoughts?

ShinobiCCTV, 2017-06-08:
Just my thoughts since there hasn't been a response yet: 
The way Id go about this is create another column named details. Make this column only a JSON string. Then put whatever values you want.

Querying inside a json is simple.
SELECT * FROM Videos Where details LIKE '%"finished":"1"%';

If you aren't into that idea you can make another column or use the executed column. In an old version of zm I customized years back I used the executed column to determine status of the videos. (Mainly because at the time I didn't know where to go look to see how this column was being used.. so I used it my way). Either way you can use this column to avoid deleting in progress events. When I did it, I used 0 to say it was building, 1 to show it's done and 2 to mark as read.

So if the row's Executed column was not 0 it would get checked over.

connortechnology, 2017-06-08:
that's a bad idea.  I might add an explicit processing column.  The JSON idea is nice too.. although may be overkill right now.

I think my specific case was the PurgeWhenFull filter not deleting events because they were in a bad state.

zmaudit.pl should probably auto-close any events that are older than a given time.

closed 2018-06-22 by stale[bot]