ZoneMinder/zoneminder #1848: API Issues with 1.30.2

conradhagemans, 2017-04-06:
Earlier this week I upgraded to 1.30.2 from 1.30 with iconners repo.

Since I can not see any streams via the API. I can see that the API user is logging in.
I tried with Home Assistant and with zmview

What has being changed? Other having same issues?

ZoneMinder Version: 1.30.2
Linux Distribution and Version: UBUNTU 14.02 LTS

ekwoodrich, 2017-04-07:
I can still view streams through zmView on 1.30.2. 

zmNinja on the other hand successfully authenticates then reports "API access failed" and won't allow me to access any streams.

ZoneMinder Version: 1.30.2
Linux Distribution and Version: Ubuntu 16.04

conradhagemans, 2017-04-07:
@ekwoodrich Hi Eliot. Thanks for the response.

I tried the URL https://myserver/zm/api/ but screens stays blank. No result at all.
So I started to check the /usr/share/zoneminder/www/api/app/tmp/cache
I found out that there was no cache folder. I created it made it writeable for www-data.
Then I checked the apache/error.log
Error missing cache/models folder and tmp/logs.
```[Sat Apr 08 01:01:00.972144 2017] [:error] [pid 25263] [client 83.161.157.241:60147] PHP Warning:  /usr/share/zoneminder/www/api/app/tmp/cache/models/ is not writable in /usr/share/zoneminder/www/api/lib/Cake/Cache/Engine/FileEngine.php on line 385```

Created them made them writeable.
errors are now gone. But:

I now get an page shown with https://myserver/zm/api/  but with error: ```"Not Authenticated" Error: The requested address '/zm/api/' was not found on this server.```

When I try to connect with ZmView from my iPhone I see an error in apache2/error.log:
``` [Sat Apr 08 01:10:16.978469 2017] [ssl:error] [pid 32652] AH02032: Hostname www.hagemans.com provided via SNI and hostname www.hagemans.com:(null) provided via HTTP are different```

I can still login to zm via browser (https) and see streams of my camera's

So since my update to 1.30.2 something messed up the settings.

conradhagemans, 2017-04-07:
Now I created all those directories manualy I checked Homeassistant again.

And.... It works!

But ZmView from my iphone not.
Homeassistant is running on the same server as Zoneminder. Maybe that why he can connect and fetch the streams.

ekwoodrich, 2017-04-08:
@conradhagemans Glad to hear you've made *some* progress at least!

That CakePHP error on `/zm/api/` doesn't seem to actually mean the API isn't working. I got the same message and I tested `/zm/api/host/getVersion.json` on the browser and it returned a result successfully. 

If the direct API functions work than it might be something on the client-side, not Zoneminder itself...

At the moment (after adding those CakePHP folders and fixing permissions/ownership, zmView is in fact working for me) with SSL on 1.30.2.

 ----
The error from your Apache log doesn't seem to be directly related to Zoneminder (necessarily)... have you tried to clear/flush your DNS cache/clearing zmView's internal cache (not sure how that works on iOS)? 

Also, I suppose you could try connecting directly to the IP rather than your hostname to rule out any DNS related issues with the SSL negotiation.

Hope something works, updating ZM and keeping your setup always seems to be a challenge!

conradhagemans, 2017-04-08:
Thanks.
The API is indeed working.
As a test I installed zmNinja on my iPhone and it works perfect.

I guess that their is a problem in the repo from Isaac Conner if you upgrade to 1.30.2
Thanks for all the info.

I will close this issue.

closed 2017-04-08 by conradhagemans