ZoneMinder/zoneminder #1865: Timestamp Label Format (%N, %f, %Q) / behaviour of strftime() on FreeBSD

schrorg, 2017-04-27:
When using ZoneMinder on FreeBSD (at least on FreeBSD 11.0), all my cameras are named "N" in the image timestamps. This is because strftime() on FreeBSD removes the "%" sign of unknown conversion specifications instead of not touching it.
On Linux, where strftime() does not touch the unknown `%N`, `%f`, and `%Q` in the LabelFormat string, camera names are printed correctly into the images.
As a workaround, putting `%%N` in the LabelFormat string of a camera makes the substitiutions work on both Linux and FreeBSD for my cameras, because strftime() than simply substitutes `%%` with `%` and so leaves `%N` in the string.

I would like to fix this but are not sure what solution is the "best". There are several possible solutions:

- add a second `%` sign to the default LabelFormat string for the camera name (make it `%%N - %d/%m/%y %H:%M:%S`) or
- use a different character for substitutions that are manually done after strftime() (i.e. `!N - %d/%m/%y %H:%M:%S`) or
- change the logic in `src/zm_monitor.cpp` to first replace ZoneMinder substitutions and than use strftime() on the LabelFormat string (this would be in `Monitor::TimestampImage()` beginning around line 3151 and could be dangerous because of user input).

Unfortunately the behaviour of strftime() in situations where it finds unknown conversion specifications is undefined, so I would prefer to change the default LabelFormat string to double the `%`. So every existing installation would work as before and new cameras on updated or new installations would have the new default LabelFormat string. Also, the documentation should be updated then.

What do you think?

Thanks for reading this longish text...
Robin

knnniggett, 2017-04-27:
My brain is telling me this was discovered, and possibly addressed previously, but right now I'm only at 1/2 out of 2 cups of coffee so that's all I got right now.

@abishai Did you recall running into this? Is this one in the list the FreeBSD patches you plan to give us?

abishai, 2017-05-03:
@knnniggett  Yes, actually I opened an issue here, however I can't find it now. I found the same as @schrorg, it was rather confusing. %%N is not the perfect solution as extra % comes to API/zmNinja for monitor name.

knnniggett, 2017-08-03:
Doesn't solve the problem, but provides some background:
https://bugs.python.org/issue9811

I keep thinking about this from time to time, trying to imagine what the best solution should be moving forward.  At the moment I'm inclined to make the substitution character configurable under Options. The default should remain "%" since that works for the majority, but we could easily tie this into cmake, so that one could change this at build time.

PR #1972 (2017-08-21): allow changing of the timestamp char code delimeter
fixes #1865 

This PR allows one to change the timestamp code character, under Options -> Config, from the default "%" to anything the user wants. Note that it only affects ZoneMinder's builtin codes. This has no affect on the standard strftime codes.

As described in #1865, this is one way to address the issue of BSD operating systems handling the % character differently than their Linux counterpart.

I'm willing to hear alternative methods to fix this.

@schrorg  @abishai 
Please test this and provide feedback as I don't currently have any  [... cut]
closed 2017-08-21 by connortechnology