ZoneMinder/zoneminder #1870: Name "Net::SFTP::Foreign::debug" used only once: possible typo at /usr/bin/zmfilter.pl line 939.

antares2001, 2017-04-30:
Hello!

On Zoneminder 1.29 (Debian Jessie), I get the following error when executing zmfilter.pl:

`Name "Net::SFTP::Foreign::debug" used only once: possible typo at /usr/bin/zmfilter.pl line 939.`

I think this is related to issue https://github.com/ZoneMinder/ZoneMinder/issues/368 but that issue has never been resolved. I have the suggested package installed via apt-get but the error still prevails.

Since I have no experience with perl, I would be very glad if someone could give a hint of how to fix this. It seems that this prevents background execution of tasks (at least I don't get anything executed in background).

Thank you!

knnniggett, 2017-04-30:
In Perl, the message "used only once...." is just a warning. The rest of the script continues to work normally so even when the user did see this message, the only affect it had was a lack of debug output.

We fixed this a year ago:
https://github.com/ZoneMinder/ZoneMinder/commit/ab3c35c629266d804c5f0c29839ddcfb9a2cceb7

Turns out there is no such thing as a Net::SFTP::Foreign::debug module, and we had to come up with a different solution.

If you no longer want to see that Perl warning, then you should upgrade to the latest version.

closed 2017-04-30 by knnniggett

antares2001, 2017-05-03:
Hello @knnniggett 

thank you for the information. Okay - then the problem with background tasks must be of a different source.

Unfortunately, as I'm running on Debian 8, there is no update to Zoneminder available, but I'll just ignore the message.

knnniggett, 2017-05-03:
Or you could build your own package from our documentation:
https://zoneminder.readthedocs.io/en/stable/installationguide/ubuntu.html#id4

If you need help with this then feel free to contact us in the user forum or our irc channel

connortechnology, 2017-05-03:
I may soon have jessie packages built on our zmrepo ppa. It will be much later today.

antares2001, 2017-05-04:
Hello @connortechnology,

I googled for that and figured it should be this one, right?

`add-apt-repository ppa:iconnor/zoneminder`

Thanks for your effort. I'll give it a try then when I get home.

connortechnology, 2017-05-04:
Um, no, that is for ubuntu.

It is taking me a while to generate debian packages... 

When ready, the line to add to /etc/apt/sources.list will look like this.
deb http://zmrepo.zoneminder.com/debian/stable	jessie/

antares2001, 2017-05-05:
Thanks a lot for the hint. I've added the line to my sources, but currently get an error on apt update, so I suppose that's still in progress. I'll give it a try later the day again.

Thanks a lot, once again! This is great news to get the newest ZoneMinder running on my system. BTW - is that also including support for mp4-based recording instead of jpegs?

antares2001, 2017-05-26:
Hello Ian,

did you manage to build a package for Debian Jessie or is that undertaking given up?

connortechnology, 2017-05-27:
Have resolved some problems with debian package building scripts.  Should be able to resume nightly builds for jessie this weekend.

Add the following to /etc/apt/sources.list for stable release packages
deb http://zmrepo.zoneminder.com/debian/release/	jessie/

antares2001, 2017-05-29:
Hello @connortechnology 

the repository works, but it only seems to contain 1.29. I'll check back and to see when 1.30 is ready. Thanks a lot for your efforts.

connortechnology, 2017-05-30:
I'm still having a lot of trouble building packages for debian.  Working on it, but it is not my primary focus. Hopefully in the next couple days it will come together.