ZoneMinder/zoneminder #1875: Enhancing/Optimizing Montage Output

nitrag, 2017-05-10:
I'm going to have more than a few users connecting to ZM and using a montage view. From what I understand Montage reads from the ZM storage and not the cameras (which would increase the connections to the cameras), this is good. However, with 5 cameras I'm seeing 250+ TCP requests per minute. What are your thoughts on having Montage be a middle-man UDP/multicast instance? This way increasing the number of viewers does not significantly impact storage I/O and CPU. With some direction I can probably implement this.

connortechnology, 2017-05-10:
The extra connections are likely due to the ajax status updates in use.  Checks for alarm status, fps, etc which go on in the background.  

I'd be interested to see how you would do a multicast... because I just don't see how that is possible in a browser context.

alfvicente, 2017-05-21:
My 2 cents: In my setup I have two separated networks, one for the cameras and the other for the workstations. They are both connected to the ZM server and excepting ZM server they can't see each other. For me to be able to connect my workstation directly to a camera stream then there should be some sort of tunneling in the server, maybe iptables or something, so I can reach the cameras.

nitrag, 2017-06-03:
@connortechnology Where is the source for `/cgi-bin/nph-zms`?

Linwood-F, 2017-06-03:
@nitrag are you looking at the Montage view, or the Montage Review?  The latter was added to provide a different mechanism for access, and it does a LOT of requests as it is a pull process rather than a push process from the server, so it manages the number of connections to work around the browser limitations (it also provides review of historical as well as live data).   I mention it because of your comment on high connection rate.

connortechnology, 2017-06-03:
Zms.cpp

nitrag, 2017-06-03:
@Linwood-F Oh man you're right! Didn't notice there was a difference.

So I like how it manages the connection workaround and I'm able to see all cameras. What bothers me is the 100% CPU utilization and 1000 TCP requests per second.

Point me in the right direction and maybe we can move this to Websocket or something.

connortechnology, 2017-07-12:
I made some improvements to montage review.  Can you comment on whether it has made a noticeable improvement for you?

Linwood-F, 2018-06-15:
FWIW I've changed products so I won't be doing further updates.

closed 2018-08-21 by stale[bot]