ZoneMinder/zoneminder #1879: After 1.30.4 update, no video or images in browser

normanhenderson, 2017-05-12:
On Ubuntu 16.04 LTS server, after update of ZM to version 1.30.4 via apt-get, it stopped displaying any images although capturing events, and the GUI otherwise appeared normal.

Found errors in log about zms not starting:
Socket /var/run/zm/zms-219340s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors()

Found that /etc/zm/zm.conf points cgi-bin files to: /var/lib/zoneminder/cgi-bin/
(ZM_PATH_CGI=/usr/lib/zoneminder/cgi-bin
)
There are 2 files there, zms and nph-zms, updated May 9. That is also the date of the latest executables.

However, /etc/apache2/conf-available/serve-cgi-bin.conf points to /usr/lib/cgi-bin/
(                ScriptAlias /cgi-bin/ /usr/lib/cgi-bin/
                <Directory "/usr/lib/cgi-bin">
)
where there are also files zms and nph-zms, older (Mar 2).

Workaround:
cd /usr/lib/cgi-bin
rm nph-zms
rm zms
ln -s /usr/lib/zoneminder/cgi-bin/nph-zms nph-zms
ln -s /usr/lib/zoneminder/cgi-bin/zms zms
service zoneminder restart

This fixed the problem, zoneminder now displays video streams.

However it looks like there was some kind of bug introduced in the update that breaks CGI.

knnniggett, 2017-05-14:
Since you are using Ubuntu, the assumption is you are using iconnor's ppa to install zoneminder.

/etc/apache2/conf-available/serve-cgi-bin.conf is the default cgi-bin folder, which zoneminder doesn't use on Ubuntu 16.04. See https://github.com/ZoneMinder/ZoneMinder/blob/master/distros/ubuntu1604/rules#L26

Instead the package installs its own cgi-bin folder, with its own apache config file /etc/apache2/conf-available/zoneminder.conf

```
abauer@Ubuntu-Desktop:~/Downloads$ dpkg -c zoneminder_1.30.4-xenial1_amd64.deb |grep conf-available
drwxr-xr-x root/root         0 2017-05-09 12:56 ./etc/apache2/conf-available/
-rw-r--r-- root/root       528 2017-05-09 12:56 ./etc/apache2/conf-available/zoneminder.conf
```

Referring to the easy-way documentation:
https://wiki.zoneminder.com/Ubuntu_Server_16.04_64-bit_with_Zoneminder_1.30.4_the_easy_way

one can simply:
`a2enconf zoneminder`

And then the cgi would have just worked. No symlinking required. 

What I can't explain with absolute certainty is why you had old zms files under your default cgi-bin folder. It looks like the old ubuntu 15.04 package did use the default cgi-bin but why the files were not deleted after the next package upgrade is still a mystery.

If there is anything we can do on our end, I think it would be to just delete the 15.04 package scripts under our distros subfolder since I don't think we are supporting that anymore. @connortechnology would have the say on that.

normanhenderson, 2017-05-14:
Thanks Andrew. Interesting – the iconnor zoneminder ppa is in fact in sources.list.d, but I had forgotten about it.

Yes, there exists a conf-available/zoneminder.conf which I hadn’t noticed either!

However the reason why I think this is a bug in the package is that (a) it worked before the version upgrade to 1.30.4; (b) an innocent user like me shouldn’t have to find documentation about, and run a2enconf zoneminder after an upgrade done via apt.

In this case the upgrade was not Ubuntu 15.04 to 16.04 LTS but simply a point-upgrade of zoneminder, using apt-get update && apt-get safe-upgrade.

Best regards, Norman

https://wiki.zoneminder.com/Ubuntu_Server_16.04_64-bit_with_Zoneminder_1.30.4_the_easy_way

knnniggett, 2017-05-14:
It is critically important to follow the documentation after each and every installation or upgrade. I don't think this is an unreasonable requirement. Feel free to disagree with that, but I think we both can agree that, had the instructions been followed, your zoneminder system would have worked without issue. 

While it might be easy to argue the one step you needed could be automated, many of the other steps cannot be, which brings us right back to following the documentation after each and every upgrade.

Looking ahead, we would love to have an install wizard, which would minimize these kinds of issues as it would draw the user's attention to potential problems like this. We do have an open issue for that, so I'm going to close this one.

closed 2017-05-14 by knnniggett

normanhenderson, 2017-05-15:
Andrew / knnniggett,

You are missing the point.

If I had downloaded Zoneminder from the site and installed from source, your comments would be valid. And, it is noted that Isaac Connor's PPA for ZM on Ubuntu is "unsupported". 

But actually I don't want an install wizard. I don't want to have to deal with ZM as a special case; we are a charity hospital in Africa and don't have resources for that. I want to be able to update all of the software on our several Linux systems through apt, in a standard way, without having to refer to release notes each time. For the same reason I want all of our Linux systems to use a single distro, and we have chosen Ubuntu because it is widely used and supported by the community.

Rather than writing an install wizard, I suggest you direct your efforts towards integrating ZM into the release management mechanisms of the major distros. Unless you want Zoneminder to be used mostly by geeks who like to, and can afford the time to solve obscure technical problems.

knnniggett, 2017-05-15:
@normanhenderson I can see your point quite clearly, and I disagree with it. Those of us who work on the project are just volunteers who do this in our spare time, and like you we don't have the resources to build and maintain a package for one user's specific requirements. 

What you are asking us to do, for free in our spare time, is to write a package that configures the database, the web server, authentication, and a host of other options to your particular requirements. You've made the assumption that the rest of our userbase has the exact same requirements as you, and that simply isn't true. There is no one-size fits all package. 

This requirement is not unique to ZoneMinder. Any application requiring a LAMP stack is going to require user intervention and any updates should be done with caution.

With that said, I do agree ZoneMinder can be complicated, and we do strive to make it easier. However, that does not mean one would can expect to not have to read the documentation. That isn't one of our goals, nor should it be.

I wish you luck at your hospital.