ZoneMinder/zoneminder #1880: cant start zoneminder

jace479, 2017-05-12:
You should only file an issue if you found a bug.  Feature and enhancement requests, general discussions and support questions should occur in one of the following areas:

- The ZoneMinder IRC channel - irc.freenode.net #zoneminder
- The [ZoneMinder Forum](https://forums.zoneminder.com/)

**Do not post feature or enhancement requests, general discussions or support questions here.**

Make sure you are running the latest version of ZoneMinder before reporting an issue.

**ZoneMinder Version (`zmaudit.pl -v`):**

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**
Ubuntu (x86-64) 16.04 LTS
**If the issue concerns a camera, provide the make, model, frame rate, resolution and ZoneMinder Source Type:**

**Relevant log lines:**
```
log lines here
```
zoneminder.service - ZoneMinder CCTV recording and surveillance system
   Loaded: loaded (/lib/systemd/system/zoneminder.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2017-05-12 10:22:54 CDT; 12s ago
  Process: 28467 ExecStart=/usr/bin/zmpkg.pl start (code=exited, status=255)

May 12 10:22:54 zornj-DV051-L zmpkg.pl[28467]: Can't connect to db at /usr/share/perl5/ZoneMinder/Config.pm line 119.
May 12 10:22:54 zornj-DV051-L zmpkg.pl[28467]: BEGIN failed--compilation aborted at /usr/share/perl5/ZoneMinder/Config.pm line 119.
May 12 10:22:54 zornj-DV051-L zmpkg.pl[28467]: Compilation failed in require at /usr/share/perl5/ZoneMinder.pm line 33.
May 12 10:22:54 zornj-DV051-L zmpkg.pl[28467]: BEGIN failed--compilation aborted at /usr/share/perl5/ZoneMinder.pm line 33.
May 12 10:22:54 zornj-DV051-L zmpkg.pl[28467]: Compilation failed in require at /usr/bin/zmpkg.pl line 48.
May 12 10:22:54 zornj-DV051-L zmpkg.pl[28467]: BEGIN failed--compilation aborted at /usr/bin/zmpkg.pl line 48.
May 12 10:22:54 zornj-DV051-L systemd[1]: zoneminder.service: Control process exited, code=exited status=255
May 12 10:22:54 zornj-DV051-L systemd[1]: Failed to start ZoneMinder CCTV recording and surveillance system.
May 12 10:22:54 zornj-DV051-L systemd[1]: zoneminder.service: Unit entered failed state.
May 12 10:22:54 zornj-DV051-L systemd[1]: zoneminder.service: Failed with result 'exit-code'.

SteveGilvarry, 2017-05-12:
Cannot connect to database, is not a bug, please use zoneminder forums for usage issues. Also follow one of the install guides to ensure you installed and setup the database. https://wiki.zoneminder.com/Ubuntu

closed 2017-05-12 by SteveGilvarry

jace479, 2017-05-15:
i recheked everything now it shows this "Cannot write to content
dirs('events','images'). Check that these exist and are owned by the web
account user"