ZoneMinder/zoneminder #1899: ZoneMinder ONVIF errors on add

aglyph, 2017-05-31:
ZoneMinder Version: Release 1.30.4
Linux Distro: Ubuntu 17.04
Camera Make & Model: Axis M2026-LE ; Firmware 6.15.4.1 (latest available FW)

Issue: 

ZoneMinder ONVIF Probe correctly detects camera through web GUI (Add New Monitor --> ONVIF --> Detected Camera M2026-LE @ 10.10.23.100); however once username/password is entered ZoneMinder responds with "Unable to probe network cameras, status is '255'.", along with strings of "ERror cannot add.." and "Nothing returned.." 

"zmonvif-probe.pl -v probe" output attached; shows M2026-LE is ONVIF 1.2 compliant and is available on http://10.10.23.100/onvif/device_service

Running "zmonvif-probe.pl -v profiles http://10.10.23.100/onvif/device_service 1.2 username password" shows similar output to the web GUI; output attached.

Wireshark of zmonvif-probe.pl profiles shows ZoneMinder with a successful POST to http://10.10.23.100/onvif/device_service ; however subsequent ZoneMinder POST to http://10.10.23.100/onvif/services receives from camera HTTP response "400 Bad Request".

Able to fully access ONVIF functions on camera using ONVIF Device Manager; wireshark shows ONVIF Device Manager POST to http://10.10.23.100/onvif/services receiving from the camera HTTP response "200 OK".

[zmprobe.txt](https://github.com/ZoneMinder/ZoneMinder/files/1039981/zmprobe.txt)
[zmprofile.txt](https://github.com/ZoneMinder/ZoneMinder/files/1039980/zmprofile.txt)

connortechnology, 2017-05-31:
Any chance we can get direct access to this camera?

aglyph, 2017-05-31:
This can be arranged; should details be sent to email in your bio or different location?

juancitotrucupe, 2017-06-30:
Hi, I am having the same problem.

ZoneMinder Version: Release 1.30.4
Linux Distro: Ubuntu 14.04
Camera Make & Model: Reolink RLC-410; Firmware v2.0.0.1218_16100801

I have two of these cameras. same issue on both. Here is the output of zmprobe:

[zmprobe.txt](https://github.com/ZoneMinder/ZoneMinder/files/1115945/zmprobe.txt)
[zmprofile.txt](https://github.com/ZoneMinder/ZoneMinder/files/1115947/zmprofile.txt)

connortechnology, 2017-07-13:
details can be sent to me at isaac@zoneminder.com

closed 2018-06-22 by stale[bot]

Kernel-Fault, 2018-07-08:
Same  thing any  other RTSP and ONVIF software work in the same LAN except ZoneMinder. Alltime getting 'Unable to probe network cameras, status is '255''.

Cramac, 2018-08-12:
same problem.
when trying to add a camera via: Add New Monitor -> ONVIF -> my cam and login/pass -> stream -> save
Nothing returns to the add monitor window.
Apache log:

> [Sun Aug 12 22:40:44.483447 2018] [php7:notice] [pid 7726] [client 93.80.246.221:57464] PHP Notice:  unserialize(): Error at offset 0 of 460 bytes in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 176, referer: http://site/zm/index.php
> [Sun Aug 12 22:40:44.483513 2018] [php7:warn] [pid 7726] [client 93.80.246.221:57464] PHP Warning:  Invalid argument supplied for foreach() in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 177, referer: http://site/zm/index.php

in this place (monitor.php on line 177), I put print_r($_request);
and this is what he will translate into the main window:
`Array ( [view] => monitor [mid] => 0 [step] => [probe] => YTo4OntzOjg6IkZ1bmN0aW9uIjtzOjc6Ik1vbml0b3IiO3M6NDoiVHlwZSI7czo2OiJGZm1wZWciO3M6NDoiSG9zdCI7czozOToiaHR0cDovLzEwLjEwLjEwLjE4L29udmlmL2RldmljZV9zZXJ2aWNlIjtzOjQ6IlNPQVAiO3M6NDoiIDEuMiI7czo0OiJOYW1lIjtzOjI0OiJISUtWSVNJT04lMjBEUy0yQ0QyMDMyLUkiO3M6NToiV2lkdGgiO3M6NDoiMTkyMCI7czo2OiJIZWlnaHQiO3M6NDoiMTA4MCI7czo0OiJQYXRoIjtzOjk2OiJydHNwOi8vYWRtaW46WW90YUxqY25AMTAuMTAuMTAuMTg6NTU0L1N0cmVhbWluZy9DaGFubmVscy8xP3RyYW5zcG9ydG1vZGU9bWNhc3QmcHJvZmlsZT1Qcm9maWxlXzEiO30= [saveBtn] => Save )`

frazi, 2018-08-26:
same problem here.