ZoneMinder/zoneminder #1910: Playback controls on event don't work

kimmeld, 2017-06-08:
**ZoneMinder Version (`zmaudit.pl -v`):**

1.31.0

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**

Latest from zmrepo-testing:  1.31.0-1.102.20170607gitg86b3b10d.el7.centos

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**

CentOS Linux release 7.3.1611 (Core)

** Issue **

When playing back an event (...index.php?view=event), the playback controls are non-functional.  When the buttons are clicked, they flash into the new state then go back into the old one.

There are no log entries related to this, however "web_php[17711]: ERR [ReferenceError: Can't find variable: barWidth]" appears in the logs repeatedly while viewing video.

Poking through the web inspector/JavaScript debugger, it seems like the CMD_PAUSE is never getting sent to the stream, it's always sending CMD_PLAY.  It seems like the pause command gets queued then getCmdResponse (event.js) calls streamPlay( false ) when it gets a status message from the stream which resets the status to playing.

** Solution **

It appears that the 'action' parameter on the various stream functions (streamPlay, streamPause, etc) is used to indicate that this was a user-initiated action vs one initiated by getCmdResponse.  I added a guard (if (action) { }) around the calls to streamReq.send and it seems to resolve the problem.

I've attached my modified [event.js](https://github.com/ZoneMinder/ZoneMinder/files/1061631/event.js.txt)

knnniggett, 2017-08-15:
Today's package from our build system contains multiple changes to event viewing.
Please try that and report back.

In particular, this commit might have solved the issue described here: 5e75588bfb85c6202dffbadcba5e4e91e3965fe9

Please

kimmeld, 2017-08-15:
I just updated my package from zmrepo-testing and the playback controls
work as expected!

Thanks for the fix!

knnniggett, 2017-08-15:
Thanks for the feedback.
It's good to see that someone is using our new rpm testing repo.

closed 2017-08-15 by knnniggett