ZoneMinder/zoneminder #1914: Issue playing back high resolution h264 events

flashoftheblades, 2017-06-10:
I'm running a pull of the master branch from 6/9/17 (1.31.0). I seem to be able to playback events from my lower resolution sub streams (640x360), but can't playback events that are triggered for the higher resolution stream (2048x1536).

I'm using Firefox and get the error **`The video could not be loaded, either because the server or network failed or because the format is not supported`**

When using IE it just spins trying to load forever.

I can see the events are being generated, as I can SCP to the server and pull down the MP4's which play fine locally. Not sure where to start looking to resolve this, whether it's an apache issue or if it's some configuration within zoneminder.

SteveGilvarry, 2017-06-22:
PHP running out of memory on large files, not sure might be memory_limit value, wasn't an issue when direct streaming from disk, but now using PHP to expose files from disk.

closed 2017-06-22 by SteveGilvarry

connortechnology, 2017-07-03:
Can you please verify that this is still an issue in current master?

flashoftheblades, 2017-07-08:
Just did a pull from the master this morning, and it appears to be working properly now. I was able to immediately start playing some of my large events (about 20min at 2K res). Skipping around seems to work as well. There doesn't seem to be the same buffering lag that there was before. Thanks!