ZoneMinder/zoneminder #1915: Dockerfile will not build

MarkosVakondios, 2017-06-11:
[100%] Built target man
-- Installing: /etc/zm.conf
CMake Error at cmake_install.cmake:60 (file):
  file INSTALL cannot find "/ZoneMinder/conf.d".

make: *** [install] Error 1
Makefile:61: recipe for target 'install' failed
The command '/bin/sh -c make && make install' returned a non-zero code: 2

knnniggett, 2017-06-11:
Thanks for spotting this. I had no idea the Docker file picks individual folders from the ZoneMinder source, rather than glob everything. Is that to cut down on filesize?

MarkosVakondios, 2017-06-11:
Can you please pull again, because there is a typo in the Dockerfile fix that will still prevent it from building successfully.
Sorry, my mistake.

knnniggett, 2017-06-11:
The original PR was already merged so it's not possible to make further changes to it. You will need to submit a new PR to correct the previous.

knnniggett, 2017-06-13:
fixed by #1915 so closing

closed 2017-06-13 by knnniggett