ZoneMinder/zoneminder #1918: Editing zones only shows black instead of color and doesn't refresh zone definitions after moving points

tschiu, 2017-06-12:
I just upgraded to 1.30.4 using the Ubuntu PPA and now when I'm editing zones in the camera:
1.  The feed is now live instead of a snapshot (yay!!!)
2.  When viewing the zones, the zones still show up as before (color coded)

[image]

3.  When clicking to edit a zone:
  3a.  The zone is black

[image]

  3b.  After moving a point, the zone doesn't update with the new zone shape (the point is a bit difficult to see -- it's the top left one)

[image]

  3c.  Clicking "save" will result in the new points being saved properly and the zone will look like the intent of the edit.

Make sure you are running the latest version of ZoneMinder before reporting an issue.

**ZoneMinder Version (`zmaudit.pl -v`):**
1.30.4
**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**
NAME="Ubuntu"
VERSION="16.04.2 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.2 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial

**If the issue concerns a camera, provide the make, model, frame rate, resolution and ZoneMinder Source Type:**
Remote / modetect / raspberry pi feeding raspi_mjpeg stream at 2592 x 1944, 5 fps approx 700k per jpg image
**Relevant log lines:**
```
log lines here
```

connortechnology, 2017-06-13:
The only thing I can think of is that the css file that gives the zone it's style isn't being loaded.  This file would be zone.css and is located at /usr/share/zoneminder/www/skins/classic/css/<?css skin name?>/views/zone.css

css skin name would be one of classic, dark or flat.  

I have just tested all css skins in 1.30.4 and they work for me.  So check your apache logs for 404's... 
Another thought is that your browser might be caching an old one.  So maybe try clearing caches.

tschiu, 2017-06-16:
Please close this issue out.  Clearing the cache didn't work (probably need a browser restart), but opening it in an incognito window did take care of it.

closed 2017-06-22 by SteveGilvarry