ZoneMinder/zoneminder #1920: Fresh Docker install, mysql permission error: Access denied for user 'zmuser'@'localhost'

marioedgar, 2017-06-13:
I've done a fresh docker install following the wiki page: https://github.com/ZoneMinder/ZoneMinder/wiki/Docker

Everything appears to have installed correctly, but when i hit the web server end point i get a mysql error:

Unable to connect to ZM db.SQLSTATE[HY000] [1045] Access denied for user 'zmuser'@'localhost' (using password: YES)

I tested this on a fresh ubuntu 16.04.2 LTS as well as my Macbook pro. I get the exact same message on both machine.

SteveGilvarry, 2017-06-22:
It seems to want to create the account here, are you seeing any other parts of the setup script missing. https://github.com/ZoneMinder/ZoneMinder/blob/master/utils/docker/setup.sh#L41

knnniggett, 2017-07-09:
closing, since the op has not responded

closed 2017-07-09 by knnniggett

konstantinj, 2017-08-10:
Hi, same issue here.

mysql get's not initialized. I guess that's because the setup.sh script checks if the /var/lib/mysql dir exists and this one existed for me already before running setup the first time.

I'll start more investigations.

konstantinj, 2017-08-10:
Problem was that there was no host directory for the mysql volume set, therefore the directory was already existing and the setup didn't run.

vegbrasil, 2017-09-05:
Hi, I'm having the same issue. Is there any fix?

Fresh container for testing, initialized with `docker run -P -d -t kylejohnson/zoneminder`

knnniggett, 2017-09-05:
Try this:
https://github.com/ZoneMinder/ZoneMinder/issues/1968#issuecomment-325980495

We are attempting to get our docker image managed by linuxserver.io
This is going to be a bit of learning curve for all of us.