ZoneMinder/zoneminder #1921: issues while saving events using videostorage

josh--, 2017-06-13:
I ran into an issue where events where not being recorded while using video storage. I would put in a pull request but i'm not a big C++ guy and not sure the "best" way to fix the issue. 

on current master the following file @ line 87 is missing a parameter where it adds the event to the DB. 
https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_event.cpp#L87

specifically I changed 
`( %d, 'New Event', from_unixtime( %ld ), %d, %d, '%s', '%s', '%d' )`
to 
`( %d, 'New Event', from_unixtime( %ld ), %d, %d, '%s', '%s', '%d', '%d' )
`
and it has been working fine since.

josh--, 2017-06-13:
also for reference here is the error that was showing up in the log
`Can't insert event: Column count doesn't match value count at row 1. sql was (insert into Events ( MonitorId, Name, StartTime, Width, Height, Cause, Notes, StateId, Videoed ) values ( 10, 'New Event', from_unixtime( 1497325700 ), 1280, 720, 'Motion', 'Motion: All', '1' ))`

connortechnology, 2017-06-14:
Thanks, fix committed.

closed 2017-06-14 by connortechnology