ZoneMinder/zoneminder #1924: Small shm causes bus error

AndrolGenhald, 2017-06-16:
Running a Docker container built from 8527e5f1de37d14abdf4c8c7feafb4fb999ebdbb

My container by default had 64MiB for shm, zm_monitor.cpp:526 tried to allocate almost 300MiB, then crashed with a bus error when zm_monitor.cpp:385 tried to zero it.

knnniggett, 2017-06-17:
If you look at our dockerfile, we call start.sh towards the end of the script:
https://github.com/ZoneMinder/ZoneMinder/blob/master/Dockerfile#L108

One of the first things that script does is set /dev/shm to 512MB:
https://github.com/ZoneMinder/ZoneMinder/blob/master/utils/docker/start.sh#L3

So we need to figure out why that isn't happening on your end.

Note that no one on the development team is currently using Docker so my answers are going to be a bit generic. For any docker related changes, we accept pull requests from anyone willing to maintain it.

knnniggett, 2017-06-29:
No response received from the op so closing this one out.

closed 2017-06-29 by knnniggett

AndrolGenhald, 2017-07-03:
Running `umount /dev/shm` causes the error `umount: /dev/shm: must be superuser to unmount`, since it's being run as root I assume docker prevents it. As far as I know the only way to set shm size is to use `--shm-size=4g` or similar when running `docker run`, although it would be nice if it were detected and logged instead of having to run it manually to discover that it's a bus error.