ZoneMinder/zoneminder #1928: Failed upgrade from 1.30.3 to 1.31.1

zhuykovkb, 2017-06-20:
Hi. I have upgrade zm from ppa:iconnor/zoneminder-master with apt-get and have an error.
```
root@e7b284594b15:/# zmupdate.pl                       

Initiating database upgrade to version 1.31.1 from version 1.30.3

Please ensure that ZoneMinder is stopped on your system prior to upgrading the database.
Press enter to continue or ctrl-C to stop : 

Do you wish to take a backup of your database prior to upgrading?
This may result in a large file in /tmp/zm if you have a lot of events.
Press 'y' for a backup or 'n' to continue : n

Upgrading database to version 1.31.1
Loading config from DBNo option 'ZM_DIR_EVENTS' found, removing.
No option 'ZM_DIR_IMAGES' found, removing.
No option 'ZM_DIR_SOUNDS' found, removing.
No option 'ZM_FRAME_SOCKET_SIZE' found, removing.
No option 'ZM_OPT_FRAME_SERVER' found, removing.
No option 'ZM_PATH_ARP' found, removing.
No option 'ZM_PATH_LOGS' found, removing.
No option 'ZM_PATH_MAP' found, removing.
No option 'ZM_PATH_SOCKS' found, removing.
No option 'ZM_PATH_SWAP' found, removing.
No option 'ZM_PATH_ZMS' found, removing.
 207 entries
Saving config to DB 207 entries
Upgrading DB to 1.30.4 from 1.30.3

Database successfully upgraded to version 1.30.4.
Upgrading DB to 1.31.0 from 1.30.3

Database successfully upgraded to version 1.31.0.
Upgrading DB to 1.31.1 from 1.30.3

Database successfully upgraded to version 1.31.1.

Database upgrade to version 1.31.1 successful.

root@e7b284594b15:/# zmaudit.pl 
Use of uninitialized value in concatenation (.) or string at /usr/bin/zmaudit.pl line 78.
Use of uninitialized value $Config{"ZM_DIR_EVENTS"} in pattern match (m//) at /usr/bin/zmaudit.pl line 79.
Use of uninitialized value in concatenation (.) or string at /usr/bin/zmaudit.pl line 79.
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/ZoneMinder/Logger.pm line 154.
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/ZoneMinder/Logger.pm line 199.
Database monitor '9' does not exist in filesystemDatabase monitor '11' does not exist in filesystemDatabase monitor '4' does not exist in filesystemDatabase monitor '6' does not exist in filesystemDatabase monitor '8' does not exist in filesystemDatabase monitor '1' does not exist in filesystemDatabase monitor '7' does not exist in filesystemDatabase monitor '5' does not exist in filesystemDatabase monitor '2' does not exist in filesystemDatabase monitor '3' does not exist in filesystemDatabase monitor '10' does not exist in filesystemUse of uninitialized value $Config{"ZM_PATH_SWAP"} in pattern match (m//) at /usr/bin/zmaudit.pl line 462.
Can't stat : No such file or directory
 at /usr/bin/zmaudit.pl line 463.
Deleted 1808 log table entries by time
root@e7b284594b15:/# lsb_release -c
Codename:       trusty
root@e7b284594b15:/# zmaudit.pl -v 
Use of uninitialized value in concatenation (.) or string at /usr/bin/zmaudit.pl line 78.
Use of uninitialized value $Config{"ZM_DIR_EVENTS"} in pattern match (m//) at /usr/bin/zmaudit.pl line 79.
Use of uninitialized value in concatenation (.) or string at /usr/bin/zmaudit.pl line 79.
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/ZoneMinder/Logger.pm line 154.
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/ZoneMinder/Logger.pm line 199.
1.31.1
root@e7b284594b15:/# 

```

knnniggett, 2017-06-21:
Please show the output of these commands:
`ls -lh /etc/zm`
`ls -lh /etc/zm/conf.d`
`cat /etc/zm/zm.conf`

zhuykovkb, 2017-06-21:
```
root@e7b284594b15:/# ls -lh /etc/zm       
total 8.0K
drwxr-xr-x 1 root root 4.0K Jun 21 01:56 conf.d
-rw-r--r-- 1 root root 1.5K Aug 16  2016 zm.conf

root@e7b284594b15:/# ls -lh /etc/zm/conf.d
total 4.0K
-rw-r--r-- 1 root root 340 Jun 21 01:55 zmcustom.conf

root@e7b284594b15:/# cat /etc/zm/zm.conf  
# ==========================================================================
#
# ZoneMinder Base Configuration, $Date$, $Revision$
#
# ==========================================================================
#
# This file is generated by 'configure'. Care should be taken if manually
# editing this file as an changes may be overwritten by subsequent configuration
# or installations.
#

# Path to installed data directory, used mostly for finding DB upgrade scripts
ZM_PATH_DATA=/usr/share/zoneminder

# Path to ZoneMinder binaries
ZM_PATH_BIN=/usr/bin

# Path to ZoneMinder libraries (none at present, for future use)
ZM_PATH_LIB=/usr/lib/x86_64-linux-gnu

# Path to ZoneMinder configuration (this file only at present)
ZM_PATH_CONF=/etc/zm

# Path to ZoneMinder web files
ZM_PATH_WEB=/usr/share/zoneminder/www

# Path to ZoneMinder cgi files
ZM_PATH_CGI=/usr/lib/zoneminder/cgi-bin

# Username and group that web daemon (httpd/apache) runs as
ZM_WEB_USER=www-data
ZM_WEB_GROUP=www-data

# ZoneMinder database type: so far only mysql is supported
ZM_DB_TYPE=mysql

# ZoneMinder database hostname or ip address
ZM_DB_HOST=localhost

# ZoneMinder database name
ZM_DB_NAME=zm

# ZoneMinder database user
ZM_DB_USER=zmuser

# ZoneMinder database password
ZM_DB_PASS=zmpass

# Do NOT set ZM_SERVER_HOST if you are not using Multi-Server
# You have been warned
#
# The name specified here must have a corresponding entry
# in the Servers tab under Options
ZM_SERVER_HOST=
root@e7b284594b15:/# 
```

also i tried to make zmcustom.conf as:
`cat /etc/zm/zm.conf  | egrep -v '^$|^#' > /etc/zm/conf.d/zmcustom.conf`
it doesn't solve this trouble

knnniggett, 2017-06-21:
Your copy of zm.conf is from the previous version of zoneminder, and that is the source of the problem.
For reference, the new zm.conf looks like this (before the `@var@` make variables are expanded):
https://github.com/ZoneMinder/ZoneMinder/blob/master/zm.conf.in

The old zm.conf should have been overwritten during the package upgrade. I will look on my end to see if I can find out why this didn't happen. Can you verify that you did not overwrite zm.conf with that old copy? Previously, this was acceptable behaviour, but moving forward any changes to the variables in zm.conf should be made under the conf.d subfolder.

zhuykovkb, 2017-06-22:
thank for your answer! i have a trouble with rewrite old zm.conf and fixed it myself.

now i have:
```
root@6948b91f1875:/# /etc/init.d/zoneminder start
Starting ZoneMinder: failure

root@6948b91f1875:/# bash -x /etc/init.d/zoneminder start
+ . /lib/lsb/init-functions
+++ run-parts --lsbsysinit --list /lib/lsb/init-functions.d
++ for hook in $(run-parts --lsbsysinit --list /lib/lsb/init-functions.d 2>/dev/null)
++ '[' -r /lib/lsb/init-functions.d/20-left-info-blocks ']'
++ . /lib/lsb/init-functions.d/20-left-info-blocks
++ for hook in $(run-parts --lsbsysinit --list /lib/lsb/init-functions.d 2>/dev/null)
++ '[' -r /lib/lsb/init-functions.d/50-ubuntu-logging ']'
++ . /lib/lsb/init-functions.d/50-ubuntu-logging
+++ LOG_DAEMON_MSG=
++ FANCYTTY=
++ '[' -e /etc/lsb-base-logging.sh ']'
++ true
+ prog=ZoneMinder
+ ZM_PATH_BIN=/usr/bin
+ RUNDIR=/var/run/zm
+ TMPDIR=/tmp/zm
+ command=/usr/bin/zmpkg.pl
+ case "$1" in
+ start
+ echo -n 'Starting ZoneMinder: '
Starting ZoneMinder: + count=0
+ '[' '!' -e /var/run/mysqld/mysqld.sock ']'
+ export TZ=:/etc/localtime
+ TZ=:/etc/localtime
+ mkdir -p /var/run/zm
+ chown www-data:www-data /var/run/zm
+ mkdir -p /tmp/zm
+ chown www-data:www-data /tmp/zm
+ /usr/bin/zmpkg.pl start
+ RETVAL=255
+ '[' 255 = 0 ']'
+ '[' 255 '!=' 0 ']'
+ echo failure
failure
+ echo

+ '[' 255 = 0 ']'
+ return 255
+ exit 255
root@6948b91f1875:/#
```

how i can debug this?

knnniggett, 2017-06-22:
There are a couple of things you can do:
- Open the web console and look in the zoneminder log to see if it logged the reason
- Try to start zoneminder directly rather than the service file and watch for additional output: `zmpkg.pl start`

knnniggett, 2017-06-22:
I have been able to duplicate this issue under the following conditions:
- manually edit zm.conf before upgrading
- upgrade to 1.31.1 
- when asked whether to overwrite zm.conf, accept the default which is "no"

If you find yourself in this situation again, answering "yes" when asked whether you want to overwrite zm.conf would avoid the problem.

I noticed the the package manager did create the new config file with the name `zm.conf.dpkg-dist` so, provided you did not delete that file, issuing `mv zm.conf.dpkg-dist zm.conf` would be one way to fix the problem after the fact.

We are discussing internally if there is anything we can do to avoid this altogether. In the meantime the above suggestions are what you can do to either avoid the problem completely or work through the problem.

Aesculapius, 2017-06-22:
Thanks for this hint. My installation was also hanging after the latest update - it couldn't start Zoneminder because it was waiting for zmdc.pl. Overwriting with the dpkg-dist file solved that.

zhuykovkb, 2017-06-22:
oh, i can't find `zm.conf.dpkg-dist`. can you paste the link to this? or tell me how i can gen this( i think over `dpkg`?)

knnniggett, 2017-06-22:
> oh, i can't find zm.conf.dpkg-dist. can you paste the link to this? or tell me how i can gen this( i think over dpkg?)

That file is generated by apt (not by zoneminder) during package installation, only if you tell it not to overwrite zm.conf. If you don't see that file under /etc/zm then reinstalling the package would be the least complicated way to solve the problem.

zhuykovkb, 2017-06-22:
thank you @knnniggett . i'll try this now. before i'll start i see:
```
root@84a62cc9cfcb:/tmp/zm# zmaudit.pl                  
Database monitor '8' does not exist in filesystemDatabase monitor '1' does not exist in filesystemDatabase monitor '9' does not exist in filesystemDatabase monitor '5' does not exist in filesystemDatabase monitor '6' does not exist in filesystemDatabase monitor '2' does not exist in filesystemDatabase monitor '10' does not exist in filesystemDatabase monitor '4' does not exist in filesystemDatabase monitor '11' does not exist in filesystemDatabase monitor '3' does not exist in filesystemDatabase monitor '7' does not exist in filesystemCan't stat @ZM_TMPDIR@: No such file or directory
 at /usr/bin/zmaudit.pl line 463.
Deleted 846 log table entries by time
root@84a62cc9cfcb:/tmp/zm# zmaudit.pl  -v
1.31.1
root@84a62cc9cfcb:/tmp/zm# grep -rn ZM_TMPDIR /etc/zm/
/etc/zm/zm.conf:68:ZM_DIR_EXPORTS=@ZM_TMPDIR@
/etc/zm/zm.conf:87:ZM_PATH_SWAP=@ZM_TMPDIR@
/etc/zm/conf.d/zmcustom.conf:20:ZM_TMPDIR=/tmp/zm
root@84a62cc9cfcb:/tmp/zm# stat /tmp/zm
  File: /tmp/zm
  Size: 4096            Blocks: 8          IO Block: 4096   directory
Device: 52h/82d Inode: 147837816   Links: 2
Access: (0755/drwxr-xr-x)  Uid: (   33/www-data)   Gid: (   33/www-data)
Access: 2017-06-22 23:50:18.802640327 +0300
Modify: 2017-06-22 23:50:18.802640327 +0300
Change: 2017-06-22 23:50:18.802640327 +0300
 Birth: -
root@84a62cc9cfcb:/tmp/zm# 
```

zhuykovkb, 2017-06-22:
thanks for all, guys! i run `zmpkg.pl` with `strace` and found variables, which need to be in `zmcustom.conf`. after paste my zm works!

closed 2017-06-22 by zhuykovkb