ZoneMinder/zoneminder #1939: /usr/lib64/libavutil.so.54: error adding symbols: DSO missing from command line error while trying to compile on Slackware

orsty3001, 2017-07-04:
I'm running Slackware 14.1 64bit and am trying to install the latest version of ZoneMinder. I'm getting this error when I try to compile. 

cd /home/20003/ZoneMinder/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/zma.dir/link.txt --verbose=1
ccache /usr/bin/c++   -Wall -D__STDC_CONSTANT_MACROS -O2   CMakeFiles/zma.dir/zma.cpp.o  -o zma -rdynamic libzm.a -lavresample -lbz2 -lrt -lz -lcurl -ljpeg -lssl -lcrypto -lpthread -lpcre -lgcrypt -lgnutls-openssl /usr/lib64/mysql/libmysqlclient.so -lx264 /usr/local/lib/libavformat.a /usr/local/lib/libavcodec.a /usr/local/lib/libavdevice.a /usr/local/lib/libavutil.a /usr/local/lib/libswscale.a /usr/local/lib/libavresample.a -lvlc -Wl,-rpath,/usr/lib64/mysql: 
/usr/lib64/gcc/x86_64-slackware-linux/5.3.0/../../../../x86_64-slackware-linux/bin/ld: libzm.a(zm_mpeg.cpp.o): undefined reference to symbol 'av_image_fill_arrays@@LIBAVUTIL_54'
/usr/lib64/libavutil.so.54: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
src/CMakeFiles/zma.dir/build.make:116: recipe for target 'src/zma' failed
make[2]: *** [src/zma] Error 1
make[2]: Leaving directory '/home/20003/ZoneMinder'
CMakeFiles/Makefile2:312: recipe for target 'src/CMakeFiles/zma.dir/all' failed
make[1]: *** [src/CMakeFiles/zma.dir/all] Error 2
make[1]: Leaving directory '/home/20003/ZoneMinder'
Makefile:130: recipe for target 'all' failed
make: *** [all] Error 2

I'm running this version of ffmpeg. 

ffmpeg version 3.3.2 Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 5.3.0 (GCC)
  configuration: --prefix=/usr
  libavutil      55. 58.100 / 55. 58.100
  libavcodec     57. 89.100 / 57. 89.100
  libavformat    57. 71.100 / 57. 71.100
  libavdevice    57.  6.100 / 57.  6.100
  libavfilter     6. 82.100 /  6. 82.100
  libswscale      4.  6.100 /  4.  6.100
  libswresample   2.  7.100 /  2.  7.100

knnniggett, 2017-07-04:
ZoneMinder builds fine for me on Fedora rawhide, using ffmpeg 3.3.2 from rpmfusion.
We need to see more context of your build to understand what is going wrong.

> ccache /usr/bin/c++ -Wall -D__STDC_CONSTANT_MACROS -O2 CMakeFiles/zma.dir/zma.cpp.o -o zma -rdynamic libzm.a -lavresample -lbz2 -lrt -lz -lcurl -ljpeg -lssl -lcrypto -lpthread -lpcre -lgcrypt [...]

This line doesn't look quite right to me. Seems you are missing several "-l" flags for the libav libraries. Is the system setting it this way or are you trying to force these flags in your cmake config???

For comparison, here the same line from my build output:
> /usr/bin/c++  -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic -Wall -D__STDC_CONSTANT_MACROS -O2  -Wl,-z,relro -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -rdynamic CMakeFiles/zma.dir/zma.cpp.o  -o zma -Wl,-rpath,/usr/lib64/mysql: libzm.a -lrt -lz -lcurl -ljpeg -lssl -lcrypto -lpthread -lpcre -lgnutls /usr/lib64/mysql/libmysqlclient.so -lavformat -lavcodec -lavdevice -lavutil -lswscale -lavresample -lvlc

orsty3001, 2017-07-04:
I haven't tried to force those flags. This is what I ran. 

`cmake ZM_WEBDIR="/home/web/zm" ZM_CGIDIR="/home/web/zm/cgi-bin" ZM_DB_PASS="my-password" CMAKE_LIBRARY_PATh="/usr/lib64" -DZM_EXTRA_LIBS="avresample;bz2" -DCMAKE_VERBOSE_MAKEFILE=ON CMAKE_PREFIX_PATH="/usr" `

knnniggett, 2017-07-05:
I did not notice this initially.
Looking at your output, it would appear you have two different versions of ffmpeg installed on your system. You've got older ffmpeg libraries w/ libavutil 54 under /usr/lib64 while the newer ffmpeg has been built from-source and installed under /usr/local/lib w/ libavutil 55. The compiler found the newer ffmpeg but the linker found the older.  

There are different ways you could go about to resolve this. Removing the older version of ffmpeg included with your system sounds easy, but package dependencies may make that problematic.

You can check your LD_LIBRARY_PATH to verify /usr/local/lib is there, and is in front of /usr/lib64.

knnniggett, 2017-08-15:
I'm going to close this out since there has been no follow-up since the last suggestion.

closed 2017-08-15 by knnniggett