ZoneMinder/zoneminder #1952: Arm Neon code fails to build on the raspberry pi 3

cmisip, 2017-07-15:
**ZoneMinder Version (`zmaudit.pl -v`):**
1.31.1

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**

b5cdc5d9af6764ac8d1b1bccee78ef34c14c3126
** I am using a fork of ZoneMinder branch motion_vectors

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

**Relevant log lines:**

make[1]: Entering directory '/mnt/DISK1/build/ZoneMinder/obj-arm-linux-gnueabihf'
make[2]: Entering directory '/mnt/DISK1/build/ZoneMinder/obj-arm-linux-gnueabihf'
make[3]: Entering directory '/mnt/DISK1/build/ZoneMinder/obj-arm-linux-gnueabihf'
make[3]: Leaving directory '/mnt/DISK1/build/ZoneMinder/obj-arm-linux-gnueabihf'
make[3]: Entering directory '/mnt/DISK1/build/ZoneMinder/obj-arm-linux-gnueabihf'
[  1%] Building CXX object src/CMakeFiles/zm.dir/zm_image.cpp.o
/mnt/DISK1/build/ZoneMinder/src/zm_image.cpp:3418:128: warning: target attribute is not supported on this machine [-Wattributes]
 void neon32_armv7_fastblend(const uint8_t* col1, const uint8_t* col2, uint8_t* result, unsigned long count, double blendpercent) {
                                                                                                                                ^
/mnt/DISK1/build/ZoneMinder/src/zm_image.cpp:3809:110: warning: target attribute is not supported on this machine [-Wattributes]
 void neon32_armv7_delta8_gray8(const uint8_t* col1, const uint8_t* col2, uint8_t* result, unsigned long count) {
                                                                                                              ^
/mnt/DISK1/build/ZoneMinder/src/zm_image.cpp:3885:131: warning: target attribute is not supported on this machine [-Wattributes]
 void neon32_armv7_delta8_rgb32(const uint8_t* col1, const uint8_t* col2, uint8_t* result, unsigned long count, uint32_t multiplier) {
                                                                                                                                   ^
{standard input}: Assembler messages:
{standard input}:364: Error: selected processor does not support ARM mode `vdup.8 q12,r12'
{standard input}:370: Error: selected processor does not support ARM mode `vrshl.u8 q8,q0,q12'
{standard input}:371: Error: selected processor does not support ARM mode `vrshl.u8 q9,q1,q12'
{standard input}:372: Error: selected processor does not support ARM mode `vrshl.u8 q10,q2,q12'
{standard input}:373: Error: selected processor does not support ARM mode `vrshl.u8 q11,q3,q12'
{standard input}:374: Error: selected processor does not support ARM mode `vrshl.u8 q4,q4,q12'
{standard input}:375: Error: selected processor does not support ARM mode `vrshl.u8 q5,q5,q12'
{standard input}:376: Error: selected processor does not support ARM mode `vrshl.u8 q6,q6,q12'
{standard input}:377: Error: selected processor does not support ARM mode `vrshl.u8 q7,q7,q12'
{standard input}:378: Error: selected FPU does not support instruction -- `vsub.i8 q4,q4,q8'
{standard input}:379: Error: selected FPU does not support instruction -- `vsub.i8 q5,q5,q9'
{standard input}:380: Error: selected FPU does not support instruction -- `vsub.i8 q6,q6,q10'
{standard input}:381: Error: selected FPU does not support instruction -- `vsub.i8 q7,q7,q11'
{standard input}:382: Error: selected FPU does not support instruction -- `vadd.i8 q4,q4,q0'
{standard input}:383: Error: selected FPU does not support instruction -- `vadd.i8 q5,q5,q1'
{standard input}:384: Error: selected FPU does not support instruction -- `vadd.i8 q6,q6,q2'
{standard input}:385: Error: selected FPU does not support instruction -- `vadd.i8 q7,q7,q3'
{standard input}:1878: Error: selected processor does not support ARM mode `vabd.u8 q0,q0,q4'
{standard input}:1879: Error: selected processor does not support ARM mode `vabd.u8 q1,q1,q5'
{standard input}:1880: Error: selected processor does not support ARM mode `vabd.u8 q2,q2,q6'
{standard input}:1881: Error: selected processor does not support ARM mode `vabd.u8 q3,q3,q7'
{standard input}:33683: Error: selected processor does not support ARM mode `vdup.32 q8,r12'
{standard input}:33689: Error: selected processor does not support ARM mode `vabd.u8 q0,q0,q4'
{standard input}:33690: Error: selected processor does not support ARM mode `vabd.u8 q1,q1,q5'
{standard input}:33691: Error: selected processor does not support ARM mode `vabd.u8 q2,q2,q6'
{standard input}:33692: Error: selected processor does not support ARM mode `vabd.u8 q3,q3,q7'
{standard input}:33693: Error: selected processor does not support ARM mode `vrshr.u8 q0,q0,#3'
{standard input}:33694: Error: selected processor does not support ARM mode `vrshr.u8 q1,q1,#3'
{standard input}:33695: Error: selected processor does not support ARM mode `vrshr.u8 q2,q2,#3'
{standard input}:33696: Error: selected processor does not support ARM mode `vrshr.u8 q3,q3,#3'
{standard input}:33697: Error: selected FPU does not support instruction -- `vmul.i8 q0,q0,q8'
{standard input}:33698: Error: selected FPU does not support instruction -- `vmul.i8 q1,q1,q8'
{standard input}:33699: Error: selected FPU does not support instruction -- `vmul.i8 q2,q2,q8'
{standard input}:33700: Error: selected FPU does not support instruction -- `vmul.i8 q3,q3,q8'
{standard input}:33701: Error: selected processor does not support ARM mode `vpadd.i8 d0,d0,d1'
{standard input}:33702: Error: selected processor does not support ARM mode `vpadd.i8 d2,d2,d3'
{standard input}:33703: Error: selected processor does not support ARM mode `vpadd.i8 d4,d4,d5'
{standard input}:33704: Error: selected processor does not support ARM mode `vpadd.i8 d6,d6,d7'
{standard input}:33705: Error: selected processor does not support ARM mode `vpadd.i8 d0,d0,d0'
{standard input}:33706: Error: selected processor does not support ARM mode `vpadd.i8 d1,d2,d2'
{standard input}:33707: Error: selected processor does not support ARM mode `vpadd.i8 d2,d4,d4'
{standard input}:33708: Error: selected processor does not support ARM mode `vpadd.i8 d3,d6,d6'
{standard input}:33709: Error: selected processor does not support ARM mode `vst4.32 {d0[0],d1[0],d2[0],d3[0]},[r2]!'
src/CMakeFiles/zm.dir/build.make:398: recipe for target 'src/CMakeFiles/zm.dir/zm_image.cpp.o' failed
make[3]: *** [src/CMakeFiles/zm.dir/zm_image.cpp.o] Error 1
make[3]: Leaving directory '/mnt/DISK1/build/ZoneMinder/obj-arm-linux-gnueabihf'
CMakeFiles/Makefile2:162: recipe for target 'src/CMakeFiles/zm.dir/all' failed
make[2]: *** [src/CMakeFiles/zm.dir/all] Error 2
make[2]: Leaving directory '/mnt/DISK1/build/ZoneMinder/obj-arm-linux-gnueabihf'
Makefile:127: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/mnt/DISK1/build/ZoneMinder/obj-arm-linux-gnueabihf'
dh_auto_build: make -j1 returned exit code 2
debian/rules:45: recipe for target 'build' failed
make: *** [build] Error 2

mastertheknife, 2017-07-19:
Hi,

What GCC version?
What do "uname -p" and "uname -s" return?
What cmake version? It is possible that cmake 3.0 or higher is needed here.

Basically, you are getting that error because you are using GCC <6.0 with AArch32 code. To compile neon code, a compiler flag (-mfpu=neon) is needed.
There is this section in CMakeLists.txt that takes care of this for AArch32 systems with GCC <6.0:
https://github.com/ZoneMinder/ZoneMinder/blob/master/CMakeLists.txt#L69
But it seems its not working right for you. First suspect would be the cmake version.

Kfir

cmisip, 2017-07-23:
gcc (Raspbian 4.9.2-10) 4.9.2

uname -p unknown
uname -s Linux

cmake version 3.6.2

Chris

knnniggett, 2017-07-23:
hmmmm.... maybe we should directly check the value of "uname -m" in cases where CMAKE_SYSTEM_PROCESSOR contains "unknown".

Looking at the cmake code makes me wonder if anyone out there on *BSD will try to build zoneminder on arm. Guess we can address that if someone else posts a new issue.

mastertheknife, 2017-07-27:
So looks like thats the issue.

uname -p should return 'armv7l' or similar on AArch32, and 'aarch64' for AArch64.
What about uname -m? maybe we should use that instead.

knnniggett, 2017-08-02:
@mastertheknife The following output is from my raspberry pi 1:
```bash
abauer@workpi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
abauer@workpi:~ $ uname -a
Linux workpi 4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 armv6l GNU/Linux
abauer@workpi:~ $ uname -p
unknown
abauer@workpi:~ $ uname -m
armv6l
```
I'd expect the uname output for pi 2 & 3 to be similar.

To resolve this issue, this is what I was thinking (completely untested):
```cmake
string(TOLOWER "${CMAKE_SYSTEM_PROCESSOR}" ZM_SYSTEM_PROC)
IF((ZM_SYSTEM_PROC STREQUAL "") OR (ZM_SYSTEM_PROC STREQUAL "unknown"))
    execute_process(COMMAND uname -m OUTPUT_VARIABLE ZM_SYSTEM_PROC ERROR_VARIABLE ZM_SYSTEM_PROC_ERR)

    # maybe make the following error checks fatal
    IF(ZM_SYSTEM_PROC_ERR)
        message(WARNING "\nAn error occurred while attempting to determine the system processor:\n${ZM_SYSTEM_PROC_ERR}")
    ENDIF(ZM_SYSTEM_PROC_ERR)
    IF(NOT ZM_SYSTEM_PROC)
       message(WARNING "\nUnable to determine the system processor. This may cause a build failure.\n")
    ENDIF(ZM_SYSTEM_PROC)

ENDIF((ZM_SYSTEM_PROC STREQUAL "") OR (ZM_SYSTEM_PROC STREQUAL "unknown"))
```

Then just modify your existing logic to use ZM_SYSTEM_PROC instead of CMAKE_SYSTEM_PROCESSOR.

cmisip, 2017-08-04:
uname -m on the RPI 3 is 

`armv7l
`

mastertheknife, 2017-08-24:
I think the problem is that gcc defaults to ARMv6 mode on Raspbian because its a distro compiled for ARMv6. Neon is an optional feature introduced in ARMv7.
The solution is probably to use "-march=armv7-a+simd" instead of "-mfpu=neon". This ensures the code will compile and work on ARMv7(A) CPUs with Neon, assuming Neon was detected.
I have ordered a RPi3, i'll be able to test this in a few weeks once it arrives.

connortechnology, 2017-08-24:
I can give ssh access to a pi3 right now if anyone wants it.  Also to a pi2.  My piB is offline due to packing and moving.

knnniggett, 2017-08-24:
I just installed Raspbian on a Raspberry Pi 2 using this image:
https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/2017-07-05-raspbian-jessie-lite.zip

```
abauer@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

abauer@raspberrypi:~ $ uname -m
armv7l

```

Once Raspian was installed, I executed the following commands to build the latest zoneminder master branch:
```
wget https://raw.githubusercontent.com/ZoneMinder/ZoneMinder/master/utils/do_debian_package.sh
chmod a+x do_debian_package.sh
sudo apt-get update
./do_debian_package.sh --snapshot=NOW --branch=master --type=local
```

The build was successful and it was noted the "-mfpu=neon" parameter was being used.

```
abauer@raspberrypi:~ $ ls *jessie*
zoneminder_1.31.1~20170824134443-jessie_armhf.build
zoneminder_1.31.1~20170824134443-jessie_armhf.changes
zoneminder_1.31.1~20170824134443-jessie_armhf.deb
zoneminder-dbg_1.31.1~20170824134443-jessie_armhf.deb
zoneminder-doc_1.31.1~20170824134443-jessie_all.deb
```
@cmisip 
Is this the fork/branch you are attempting to build from?
https://github.com/cmisip/ZoneMinder/tree/motion_vectors

I noticed that this fork/branch does not contain commit 613bbfb65bb18206542f5dba50e618d70d5e94c3

cmisip, 2017-08-29:
My last attempt was based on ZoneMinder master, not the fork.  I will try it again this weekend and see what happens.    
Thanks,
Chris

cmisip, 2017-09-03:
The dobuild way of creating the packages works.  If I manually do a "dpkg-buildpackage -rfakeroot -us -uc -b" on the source downloaded by dobuild, it works.  But cloning Zoneminder, checking out master and using "dpkg-buildpackage -rfakeroot -us -uc -b " yields the error.   

Chris

cmisip, 2017-09-03:
Here are the steps that reproduce the error:
`git clone https://github.com/ZoneMinder/ZoneMinder.git`
`cd Z*`
`git checkout master`
Already on 'master'
Your branch is up-to-date with 'origin/master'.
`git submodule update --init --recursive`
`cp -rf distros/debian ./`
`dpkg-buildpackage -us -uc -rfakeroot  -b -d`

PR #1966 (2017-08-21): arm neon-check "uname -m" if CMAKE_SYSTEM_PROCESSOR is not what we expect
fixes #1952 

@mastertheknife you around? Would like your feedback

@cmisip can you test this and confirm ZoneMinder builds now on Raspbian?
closed 2018-06-22 by stale[bot]