ZoneMinder/zoneminder #1960: Fails to build with GCC 7 (error: call of overloaded 'abs(unsigned int)' is ambiguous)

lamby, 2017-08-07:
```
/tmp/buildd/zoneminder-1.30.4+dfsg/src/zm_image.cpp: In static member function 'static Image* Image::Highlight(unsigned int, Image**, Rgb, Rgb)':
/tmp/buildd/zoneminder-1.30.4+dfsg/src/zm_image.cpp:1662:57: error: call of overloaded 'abs(unsigned int)' is ambiguous
         if ( (unsigned)abs((*psrc)-RGB_VAL(ref_colour,c)) >= RGB_VAL(threshold,c) )
                                                         ^
In file included from /usr/include/c++/7/cstdlib:75:0,
                 from /usr/include/c++/7/ext/string_conversions.h:41,
                 from /usr/include/c++/7/bits/basic_string.h:6159,
                 from /usr/include/c++/7/string:52,
                 from /tmp/buildd/zoneminder-1.30.4+dfsg/dbuild/src/zm_config.h:26,
                 from /tmp/buildd/zoneminder-1.30.4+dfsg/src/zm.h:27,
                 from /tmp/buildd/zoneminder-1.30.4+dfsg/src/zm_image.cpp:19:
/usr/include/stdlib.h:735:12: note: candidate: int abs(int)
 extern int abs (int __x) __THROW __attribute__ ((__const__)) __wur;
            ^~~
In file included from /usr/include/c++/7/cstdlib:77:0,
                 from /usr/include/c++/7/ext/string_conversions.h:41,
                 from /usr/include/c++/7/bits/basic_string.h:6159,
                 from /usr/include/c++/7/string:52,
                 from /tmp/buildd/zoneminder-1.30.4+dfsg/dbuild/src/zm_config.h:26,
                 from /tmp/buildd/zoneminder-1.30.4+dfsg/src/zm.h:27,
                 from /tmp/buildd/zoneminder-1.30.4+dfsg/src/zm_image.cpp:19:
/usr/include/c++/7/bits/std_abs.h:56:3: note: candidate: long int std::abs(long int)
   abs(long __i) { return __builtin_labs(__i); }
   ^~~
/usr/include/c++/7/bits/std_abs.h:61:3: note: candidate: long long int std::abs(long long int)
   abs(long long __x) { return __builtin_llabs (__x); }
   ^~~
/usr/include/c++/7/bits/std_abs.h:70:3: note: candidate: constexpr double std::abs(double)
   abs(double __x)
   ^~~
/usr/include/c++/7/bits/std_abs.h:74:3: note: candidate: constexpr float std::abs(float)
   abs(float __x)
   ^~~
/usr/include/c++/7/bits/std_abs.h:78:3: note: candidate: constexpr long double std::abs(long double)
   abs(long double __x)
   ^~~
/usr/include/c++/7/bits/std_abs.h:84:3: note: candidate: constexpr __int128 std::abs(__int128)
   abs(__GLIBCXX_TYPE_INT_N_0 __x) { return __x >= 0 ? __x : -__x; }
   ^~~
/usr/include/c++/7/bits/std_abs.h:102:3: note: candidate: constexpr __float128 std::abs(__float128)
   abs(__float128 __x)
   ^~~
make[3]: *** [src/CMakeFiles/zm.dir/zm_image.cpp.o] Error 1
```

knnniggett, 2017-08-14:
This was fixed already in commit https://github.com/ZoneMinder/ZoneMinder/commit/97380f009b0e6a8bb6ee4ecbf0436045342cdf1e

Please try again using the latest master branch.

knnniggett, 2017-08-17:
closing, since current master builds against gcc 7.

closed 2017-08-17 by knnniggett

lamby, 2017-08-17:
Confirmed that backporting 97380f0 fixes FTBFS w/gcc 7 :)

knnniggett, 2017-08-17:
Glad to hear it. Our build system is cranking out fedora 26 rpms built with gcc 7 so I knew we fixed this along the way, following the 1.30.4 release. Hope to see a newer version of zoneminder appear in the Debian repos, if you happen to be involved with that. :-)

lamby, 2017-08-18:
Latest released version is in ``unstable``. :)