ZoneMinder/zoneminder #1963: DOC: API "Add a monitor" example needs improvement

perryh2, 2017-08-13:
This is a doc issue, first seen in an earlier version but still present in HEAD.
(The fix is easy; I intend to send a pull request shortly.)

If the API "Add a monitor" example is followed exactly, it creates an invalid definition because it includes a trailing blank in all but the last field and some fields do not allow that.

A case can be made that the API should either strip trailing blanks, or return an error and not create the entry -- and a case can also be made for including in the API doc a complete list of the field names that can be used, e.g. the GUI "Source Type" field is called simply "Type" in the API -- but I am not sufficiently knowledgeable of either ZoneMinder or its implementation languages to address either of those.  What I can do is provide a patch to the doc so that newbies like me are not led astray regarding the trailing blanks :) and illustrating the setting of the Type.

**ZoneMinder Version (`zmaudit.pl -v`):**
1.31.1
**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**
edd66dd0cf277a80bf221b867703707d87cd56d6
**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**
not relevant (doc issue).  I believe the ZoneMinder instance where I saw this is running on Ubuntu, but I don't have command-line access to the system so as to report the exact version.
**If the issue concerns a camera, provide the make, model, frame rate, resolution and ZoneMinder Source Type:**
n/a
**Relevant log lines:**
```
n/a
```

Rorosha, 2018-03-03:
Here is a PR fixing this.
https://github.com/ZoneMinder/zoneminder/pull/2048

knnniggett, 2018-03-03:
Closing as #2048 was merged

closed 2018-03-03 by knnniggett