ZoneMinder/zoneminder #1967: Zoneminder & Fedora - Path ZMS

dylanduffy, 2017-08-15:
Hi,

There is some sort of issue with Zoneminder 1.30.4 and Fedora.

The ZMS/CGI locations don't seem to work no matter what you enter.

`getStreamCmdResponse stream error: Socket /var/lib/zoneminder/sock/zms-624179s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors()` 

I hope someone can try work out what is happening.

kingofkya, 2017-08-15:
this seam like more of a forum question....

but like it says go look at your logs something is setup wrong or your cgi /apaache install has some other issue.

dylanduffy, 2017-08-15:
Hi,

This is not an isolated issue while using Fedora, I know of a few users with the same issue on a fresh install.

kingofkya, 2017-08-15:
Also witch fedora your running

dylanduffy, 2017-08-15:
We are all on Fedora release 25 (Twenty Five)

kingofkya, 2017-08-15:
I will see if i can replicate, version of package/repo its from. 

Also in the mean time if it a non production machine try  SElinux / putting it in permisive mode.

dylanduffy, 2017-08-15:
Running default v1.30.4 FYI - I will try that and get back to you

knnniggett, 2017-08-15:
This is not a software bug.
To get help, create a thread in the user forum stating the version of Fedora 25, where you got zoneminder from, the value of PATH_ZMS and the contents of your ZoneMinder apache config file.

As always make sure you follow the contents of the README.

closed 2017-08-15 by knnniggett