ZoneMinder/zoneminder #1968: Docker issues

phdelodder, 2017-08-16:
To get the docker container fully working I had to install additional packages in the container: 

For 'zmtelemetry.pl' 
Was receiving this error: exited abnormally, exit status 9
Solved it by: apt-get update; apt-get install libsys-cpu-perl  libsys-meminfo-perl 

For 'API'
Was receiving this error: 

> [Wed Aug 16 05:28:02.228577 2017] [:error] [pid 463] [client 172.17.0.1:58484] PHP Fatal error:  Uncaught Error: Call to undefined function apc_fetch() in /usr/local/share/zoneminder/www/api/lib/Cake/Cache/Engine/ApcEngine.php:78\nStack trace:\n#0 /usr/local/share/zoneminder/www/api/lib/Cake/Cache/Cache.php(363): ApcEngine->read('myapp_cake_core...')\n#1 /usr/local/share/zoneminder/www/api/lib/Cake/I18n/I18n.php(235): Cache::read('cake_dev_en', '_cake_core_')\n#2 /usr/local/share/zoneminder/www/api/lib/Cake/basics.php(636): I18n::translate('Cache engine "%...', NULL, 'cake_dev')\n#3 /usr/local/share/zoneminder/www/api/lib/Cake/Cache/Cache.php(184): __d('cake_dev', 'Cache engine "%...', '_cake_core_')\n#4 /usr/local/share/zoneminder/www/api/lib/Cake/Cache/Cache.php(151): Cache::_buildEngine('_cake_core_')\n#5 /usr/local/share/zoneminder/www/api/app/Config/core.php(375): Cache::config('_cake_core_', Array)\n#6 /usr/local/share/zoneminder/www/api/lib/Cake/Core/Configure.php(72): include('/usr/local/shar...')\n#7 /usr/local/share/zoneminder/www/api/lib/Cake/bootstrap.php(434): Configure::bootstrap(true)\n#8 /usr/local in /usr/local/share/zoneminder/www/api/lib/Cake/Cache/Engine/ApcEngine.php on line 78

Solved it: 
echo "deb http://ftp.de.debian.org/debian sid main" >> /etc/apt/sources.list
apt-get update; apt-get install php-apcu-bc

Issue ONVIF Probe when adding monitor
solved: 
apt-get update; apt-get install libexpat1-dev
cpan install Data::UUID
cpan install Class::Std::Fast
cpan install Data::Dump
cpan install SOAP::WSDL::Transport::HTTP
cpan install IO::Socket::Multicast

knnniggett, 2017-08-16:
Those of us still active with the project are not using Docker so there is currently a knowledge gap. 

What docker container are you using? This one?
https://hub.docker.com/r/kylejohnson/zoneminder/

phdelodder, 2017-08-16:
Yes that's the one i'm using.

knnniggett, 2017-08-16:
Unfortunately, Kyle has not been active with the project and he holds the keys to that container.
We can add this to the list of things we have asked him to give us access to, but we know this won't happen quickly.

We, of course, do have control over the config files in our github repo, but from my limited knowledge of Docker, the root cause of what you have described is in the container itself.

I think we ought to head in a different direction....
A few weeks back, I had a conversation with linxserver.io to get ZoneMinder hosted there.
ZoneMinder does not show in their list, but it is on their github site:
https://github.com/linuxserver/docker-zoneminder

What we could really use is someone willing to champion this and get it going.
I think hosting ZoneMinder there is a much better solution than what we have currently. Your feedback is appreciated.

phdelodder, 2017-08-16:
I would love a docker container maintained by linuxserver.io. I have several containers running created/maintained by linuxserver.io and I have nothing but kind words for his work. 

Any idea on a timeline?

knnniggett, 2017-08-16:
No, I don't, unfortunately. I'm not sure what the next step is exactly.

kylejohnson, 2017-08-17:
I'm confused.  Wouldn't it just be our [Dockerfile](https://github.com/ZoneMinder/ZoneMinder/blob/master/Dockerfile) that needs your updates?

dlandon, 2017-08-28:
I've implemented a Docker that I actively maintain here:

https://hub.docker.com/r/dlandon/zoneminder/

phdelodder, 2017-08-29:
@dlandon In description it mentions unRAID, is it also compatible on eg docker on arch linux?

dlandon, 2017-08-29:
Yes.  It was created for unRAID, but can run on any docker system.

knnniggett, 2017-08-30:
FYI, we recently had zonemidner.io host a zoneminder docker container:
https://hub.docker.com/r/linuxserver/zoneminder/

We would love it if you could test that and provide feedback on the container and provide feedback here:
https://github.com/linuxserver/docker-zoneminder

Feedback is welcome.
We are doing this to let those who know Docker better than us manage the container, and at the same time let us focus more on managing the ZoneMinder source code.

phdelodder, 2017-09-01:
@knnniggett I'm unable to watch a live feed with the docker image. I already created a ticket for https://github.com/linuxserver/docker-zoneminder/issues/4

knnniggett, 2018-01-14:
We've moved our dockerfiles here:
https://github.com/ZoneMinder/zmdockerfiles

We will continue to troubleshoot docker related issue there.

closed 2018-01-14 by knnniggett