ZoneMinder/zoneminder #1973: H264/HEVC support

connortechnology, 2017-08-22:
Putting this here for discussion and bounty.

I played with an H265 stream today, here is where we are and where we need to get to in order to say we support H265.
1. h265 currently works with our h264 passthrough code and gets stored nicely in the .mp4 just fine.
2. Not many browsers will play an h265 stream/file at this time.  Mine doesn't.  This means that we need to have zms decode the .mp4 into an mjpeg for viewing old style.  (or transcode to h264 or whatever)
3. We need to be able to detect what the browser can do I guess so that we can choose to stream the h265 or view it as mjpeg.

connortechnology, 2017-09-11:
I now have zms in my branch doing the decode from the mp4 to a jpeg stream, which is good and makes h265 support work.  Still need number 3.

ziotibia81, 2017-10-27:
A "simple" method to detect streams supported by browser could be Modernizr library.
An example [--> here <--](http://diveinto.html5doctor.com/detect.html#video-formats)