ZoneMinder/zoneminder #1974: [Docker] Can not export a event as zip

senden9, 2017-08-26:
Hi,
If I use the provided docker image I can not export a event as ZIP because the `zip` package is not installed.

ZoneMinder Version: 1.31.1

Linux Distribution and Version: Ubuntu 16.04.3 LTS

Relevant log lines:
```
[Sat Aug 26 15:08:34.165132 2017] [:error] [pid 8129] [client 172.17.0.1:35180] ERR [Command 'cat '/var/tmp/zm/zmFileList.txt' | zip -q '/var/tmp/zm/zmExport.zip' -@' returned with status 127], referer: http://127.0.0.1:1080/index.php?view=export&eids[]=1
[Sat Aug 26 15:08:34.165403 2017] [:error] [pid 8129] [client 172.17.0.1:35180] PHP Notice:  Undefined offset: 0 in /usr/local/share/zoneminder/www/skins/classic/includes/export_functions.php on line 938, referer: http://127.0.0.1:1080/index.php?view=export&eids[]=1
[Sat Aug 26 15:08:34.207127 2017] [:error] [pid 8129] [client 172.17.0.1:35180] ERR [Export Failed], referer: http://127.0.0.1:1080/index.php?view=export&eids[]=1

```

knnniggett, 2017-08-30:
@senden9 We recently had zonemidner.io host a zoneminder docker container:
https://hub.docker.com/r/linuxserver/zoneminder/

We would love it if you could test that and provide feedback on the container and provide feedback here:
https://github.com/linuxserver/docker-zoneminder

We are doing this to let those who know Docker better than us manage the container, and at the same time let us focus more on managing the ZoneMinder source code.

closed 2017-08-30 by knnniggett

senden9, 2017-08-31:
Thanks for the information. I will look at the other repository in the next few days.

commit 35c23f5f (2017-08-30): Add zip package to the docker image. Fix #1974

Otherwise no export of a event as ZIP file is possible.
commit 553482ab (2017-08-30): Merge pull request #1975 from senden9/patch-1

Add zip package to the docker image. Fix #1974