ZoneMinder/zoneminder #1976: ERR Terminating, last frame sent time 1504080096.263177 secs more than maximum of 10.000000

AndyFTP, 2017-08-30:
Hi, 

I have this error, which says last frame sent time was 1504080096.263177 secs more than maximum of 10.000000, this means the last frame time was sent at 1970-01-01, if this is not a bug, how can i fix this?

**ZoneMinder Version (`zmaudit.pl -v`):**
v1.30.4

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**
NAME="Ubuntu"
VERSION="16.04.3 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.3 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial

**If the issue concerns a camera, provide the make, model, frame rate, resolution and ZoneMinder Source Type:**
http://maxxter.biz/item.aspx?id=9357

**Relevant log lines:**
```
2017-08-30 10:11:09.041619 web_php[2121].ERR-/usr/share/zoneminder/www/includes/functions.php/2033 [Socket /var/run/zm/zms-527668s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
2017-08-30 10:11:06.545499 zms[2662].ERR-zm_monitor.cpp/4213 [Terminating, last frame sent time 1504080666.045346 secs more than maximum of 10.000000]
2017-08-30 10:11:06.006661 web_js[2121].ERR-?view=watch [getStreamCmdResponse stream error: Socket /var/run/zm/zms-527668s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors()]
2017-08-30 10:11:05.956360 web_php[2130].ERR-/usr/share/zoneminder/www/includes/functions.php/2033 [Socket /var/run/zm/zms-527668s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
2017-08-30 10:11:03.509139 zms[2661].ERR-zm_monitor.cpp/4213 [Terminating, last frame sent time 1504080663.008294 secs more than maximum of 10.000000]
2017-08-30 10:11:02.947657 web_js[2130].ERR-?view=watch [getStreamCmdResponse stream error: socket_bind( /var/run/zm/zms-527668w.sock ) failed: Address already in use - checkStreamForErrors()]
2017-08-30 10:11:02.922504 web_php[1187].ERR-/usr/share/zoneminder/www/includes/functions.php/2033 [Socket /var/run/zm/zms-527668s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
2017-08-30 10:11:02.880265 web_php[2130].ERR-/usr/share/zoneminder/www/includes/functions.php/2033 [socket_bind( /var/run/zm/zms-527668w.sock ) failed: Address already in use]
2017-08-30 10:11:00.434502 zms[2660].ERR-zm_monitor.cpp/4213 [Terminating, last frame sent time 1504080659.927604 secs more than maximum of 10.000000]
2017-08-30 10:10:59.987231 web_php[1184].ERR-/usr/share/zoneminder/www/includes/functions.php/2033 [Timed out waiting for msg /var/run/zm/zms-527668s.sock]
2017-08-30 10:10:59.895290 web_js[1187].ERR-?view=watch [getStreamCmdResponse stream error: Socket /var/run/zm/zms-527668s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors()]
2017-08-30 10:10:59.880723 web_php[1186].ERR-/usr/share/zoneminder/www/includes/functions.php/2033 [Socket /var/run/zm/zms-527668s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
2017-08-30 10:10:57.392029 zms[2659].ERR-zm_monitor.cpp/4213 [Terminating, last frame sent time 1504080656.890793 secs more than maximum of 10.000000]
2017-08-30 10:10:56.859095 web_js[1186].ERR-?view=watch [getStreamCmdResponse stream error: socket_bind( /var/run/zm/zms-527668w.sock ) failed: Address already in use - checkStreamForErrors()]
2017-08-30 10:10:56.812594 web_php[1186].ERR-/usr/share/zoneminder/www/includes/functions.php/2033 [socket_bind( /var/run/zm/zms-527668w.sock ) failed: Address already in use]
2017-08-30 10:10:54.734656 zms[2656].ERR-zm_monitor.cpp/4213 [Terminating, last frame sent time 1504080654.233988 secs more than maximum of 10.000000]
2017-08-30 10:10:54.194076 web_js[1184].ERR-?view=watch [getStreamCmdResponse stream error: Socket /var/run/zm/zms-527668s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors()]
2017-08-30 10:10:53.821892 web_php[1186].ERR-/usr/share/zoneminder/www/includes/functions.php/2033 [Socket /var/run/zm/zms-527668s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
2017-08-30 10:10:51.192770 zms[2649].ERR-zm_monitor.cpp/4213 [Terminating, last frame sent time 1504080650.692405 secs more than maximum of 10.000000]
2017-08-30 10:10:50.659342 web_js[1186].ERR-?view=watch [getStreamCmdResponse stream error: socket_bind( /var/run/zm/zms-527668w.sock ) failed: Address already in use - checkStreamForErrors()]
[... cut ...] cs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
2017-08-30 10:08:37.195682 zms[2451].ERR-zm_monitor.cpp/4213 [Terminating, last frame sent time 1504080516.695175 secs more than maximum of 10.000000]
2017-08-30 10:08:36.661700 web_js[2128].ERR-?view=watch [getStreamCmdResponse stream error: socket_bind( /var/run/zm/zms-527668w.sock ) failed: Address already in use - checkStreamForErrors()]
```

Any ideas how to fix this?

kingofkya, 2017-08-30:
This probbly should be reposted on the forum and closed on bug tracker.

You likely have a full or failing hardisk, or a crashing or corrupt mysql db. Or some other hardware failure, in my experience. I think your likely just out of room on / though.

AndyFTP, 2017-08-30:
I don't think hardware is a cause, since my system is fine and i am testing this software in virtualbox. It's also a fresh install and the mysql db runs fine, so i don't think anything is corrupt. I will try to reinstall this evening when i have the time and see if i get the same error.

knnniggett, 2017-08-30:
This output is typical of a configuration error. It is not a bug.

Our github forum is reserved for bug reports only. If there is any doubt then one should start in the user forum. This keeps our github forum relatively clean with topics that can be resolved by changing the source code, rather than troubleshooting the end user's system.

For help on this issue, create a topic in the user forum, and make sure to show how you have configured the camera in question (in zoneminder).

closed 2017-08-30 by knnniggett