ZoneMinder/zoneminder #1977: Unable to access monitor configuration

kimmeld, 2017-09-03:
When I try to access the monitor configuration (clicking the source link on the console) I get a window with a blank page.  I just updated ZoneMinder from zmrepo-testing and the problem still occurs.

**ZoneMinder Version (`zmaudit.pl -v`):**

1.31.1

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**

CentOS Linux release 7.3.1611 (Core)

**If the issue concerns a camera, provide the make, model, frame rate, resolution and ZoneMinder Source Type:**

N/A

**Relevant log lines:**
/var/log/httpd/ssl_error_log
```
[Sun Sep 03 16:30:35.255648 2017] [:error] [pid 10793] [client ****:****] PHP Fatal error:  Can't use method return value in write context in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 499, referer: https://****/zm/index.php
```

Since this is a PHP error:

```
[root@zoneminder httpd]# rpm -q php
php-5.4.16-42.el7.x86_64
```

[This question on StackOverflow](https://stackoverflow.com/questions/1075534/cant-use-method-return-value-in-write-context) seems to point to what the problem is with the indication that it's fixed in PHP 5.5.  This construct (calling empty() on a function result) seems to be fairly prevalent in monitor.php, but a quick grep doesn't seem to show it in other files.

I'm not sure what the correct resolution to this issue is.  Some potentials I see are:
- Making Zoneminder depend on PHP 5.5+, which would prevent it from working on CentOS systems.  I assume that Fedora would be fine since it has newer packages, but I haven't sparked up a Fedora VM to check this.
- Adjust the Zoneminder source to work on PHP 5.4, which would allow continued support of CentOS.

knnniggett, 2017-09-04:
While el6 w/ php 5.3 is on its way out, we do want to maintain compatibility with php 5.4, since that is what ships with el7.

Looks like this was from #1958 
@connortechnology you mind taking a look?

knnniggett, 2017-09-08:
@kimmeld I have yet to hear back from @connortechnology but it looks like the following change should work.

Try changing the portion of line 499 of /usr/share/zoneminder/www/skins/classic/views/monitor.php from `if ( !empty($monitor->Id()) )` to `if ( !$monitor->Id() )` then report back. 

Monitor->Id defaults to 0 and should never be undefined so I think we can get away with not using empty.

kimmeld, 2017-09-08:
Hi @knnniggett,

The `empty($monitor->something())` pattern is used throughout the file, so I've gone and changed all of the instances of this.  I need to revisit each usage to make sure that I inverted the logic properly - I believe that `!empty($monitor->Id())` should become `$monitor->Id()`.  I took a look at the code and it doesn't seem like there is any situation where $monitor is not an initialized Monitor object, so this is probably a reasonable approach.

After changing all of these everything seems to work as expected.

Thanks!

kimmeld, 2017-09-08:
Diff showing my changes attached.

[monitor.php.diff.txt](https://github.com/ZoneMinder/ZoneMinder/files/1288880/monitor.php.diff.txt)

knnniggett, 2017-09-08:
Thanks for the help. I've been doing all my development in fedora recently and updating my centos 7 vm to a recent build of the master branch is still on my to-do list.

Yeah, there are multiple places that need updating. I was just looking for a proof of concept so I know how to proceed. The other way to do it is to assign $monitor->Id() to a variable then used that variable throughout.

@connortechnology is involved in a move right now so he unavailable for the time being.

PR #1980 (2017-09-11): monitor.php - remove checks for empty and just check for true
fixes #1977 
Restores compatibility with php 5.4

The following properties all default to values which php considers false:
$monitor->Id
$monitor->Protocol
$monitor->Controllable
$monitor->Exif
$monitor->TrackMotion
$monitor->RecordAudio
$monitor->RTSPDescribe
$monitor->V4LMultiBuffer
$monitor->LinkedMonitors

Consequently, I think it is safe to just check for true/false in these cases rather than use the empty function.

Alternatively, we can create variables to hold the contents of each of these properties and then check if the va [... cut]
closed 2017-09-11 by connortechnology