ZoneMinder/zoneminder #1979: Add New Monitor button not working in Chrome

gus-g, 2017-09-06:
Add New Monitor button doesn't seem to work in Chrome since update, did work with v1.29. However, does appear to work under MS Edge

- Recently added PPA and updated from 1.29 to ZoneMinder Version 1.30.4 on Ubuntu 16.04.3 LTS

- Microsoft Edge 38.14393.1066.0

- Google Chrome | 60.0.3112.113 (Official Build) (64-bit) (cohort: Stable)
Revision | 95c454326a7a3153e984e50a4719924968490717-refs/branch-heads/3112@{#744}
OS | Windows
JavaScript | V8 6.0.286.56
Flash | 26.0.0.151 C:\Users\*\AppData\Local\Google\Chrome\UserData\PepperFlash\26.0.0.151\pepflashplayer.dll
User Agent | Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/60.0.3112.113 Safari/537.36

Cheers,

Gus

ke0eyh, 2017-10-23:
I have the exact same issue. 

I added ZoneMinder ppa:iconnor/zoneminder yesterday, this morning it up dated to v1.30.4 and now the "Add New Monitor" button does not work. Linux Mint 18.2 | Sonya | Google Chrome Version 61.0.3163.100 (Official Build) Built on Ubuntu , running on LinuxMint 18.2 (64-bit)

I hope it is easily resolved!
ke0eyh

closed 2018-06-22 by stale[bot]

connortechnology, 2018-09-05:
Which ppa.  Which version.