ZoneMinder/zoneminder #1983: Documentation Correction -- UBUNTU 16.04 Installation

AdamDanischewski, 2017-09-10:
It worked well yet I had a few issues, first I am not that familiar with `tasksel` to know that its not userland. 

The part where it is states: 

> Step 1: Either run commands in this install using sudo or use the below to become root

should be moved up before the part where it states: 

> tasksel install lamp-server

Afterward, everything worked flawless except that the HINT of modifying the zoneminder web interface -->Options->Paths:  
`PATH_ZMS` from  `/cgi-bin/nph-zms` to `/zm/cgi-bin/nph-zms`

Does not mention that if you don't the monitors will not display images, also I can't imagine anyone not needing to do this so instead of putting it in a HINT at the bottom of the 14.04 Installation, it should probably be included in the normal instructions. The HINT should be modified to describe the symptom[s] and then what to double check. 

I'm not trying to nitpick here, only to help further polish this well written and well functioning software.

gerdesj, 2017-09-25:
I've updated the docs with this change. See https://wiki.zoneminder.com/Ubuntu_Server_16.04_64-bit_with_Zoneminder_1.30.4_the_easy_way?title=Ubuntu_Server_16.04_64-bit_with_Zoneminder_1.30.4_the_easy_way&type=revision&diff=14818&oldid=14817

AdamDanischewski, 2017-09-26:
I'm not sure of the location of that file within the links from the home page, the link I navigated to doesn't have the old text, nor the new one (though maybe it didn't get pushed out yet). 

Here is the link, I was referring to - it is the link you find from the home page clicking on Documentation (under the search bar)-->Installation Guide-->Easy Way: Ubuntu 16.04:  
http://zoneminder.readthedocs.io/en/latest/installationguide/ubuntu.html#easy-way-ubuntu-16-04

I think it would be useful to include at the end of the server specific instructions, within or before Step 11, where it says: 

> Step 11: Making sure ZoneMinder works
>   ... 
>     Open up a browser and go to http://hostname_or_ip/zm - should bring up ZoneMinder Console
> ... 
> Congratulations Your installation is complete
> 
> PPA install may need some tweaking of ZMS_PATH in ZoneMinder options. Socket_sendto or no live streaming

closed 2017-09-28 by connortechnology

AdamDanischewski, 2017-10-01:
Closed? I see the `sudo tasksel` update, yet the most critical update for users to be able to view the camera output on the monitors still requires a change PATH_ZMS from /cgi-bin/nph-zms to /zm/cgi-bin/nph-zms is still not clear imo. 

This probably shouldn't really be required, this settings update should probably be detected, otherwise if you can't or don't want to detect the proper settings then some configuration finishing scripts could be set up for people who do fix this stuff, can upload them someplace so the user doesn't have to worry about multi-manual steps.

SteveGilvarry, 2017-10-01:
Two issues in one, so one is certainly closed.
I consider that second issue the realm of packaging, ie needs to be fixed there as then we don't need the hint at all, which I think you are saying also. Only exception would be those installing in root folder of web server or somewhere else, who are not the users to need the hint. Assuming you used the PPA so we can ask @connortechnology if PPA is setting path_zms correctly?

commit d83b9ce6 (2017-09-28): Fixes #1983 (#1989)

Fair point here we put a command it should run, but I have added assistance with this part of the install is not zoneminder specific