ZoneMinder/zoneminder #1985: HTTP API does not work

Veles, 2017-09-19:
You should only file an issue if you found a bug.  Feature and enhancement requests, general discussions and support questions should occur in one of the following areas:

- The ZoneMinder IRC channel - irc.freenode.net #zoneminder
- The [ZoneMinder Forum](https://forums.zoneminder.com/)

**Do not post feature or enhancement requests, general discussions or support questions here.**

Make sure you are running the latest version of ZoneMinder before reporting an issue.

**ZoneMinder Version (`zmaudit.pl -v`):**

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**

**If the issue concerns a camera, provide the make, model, frame rate, resolution and ZoneMinder Source Type:**

**Relevant log lines:**
```
log lines here
```

mnoorenberghe, 2017-09-21:
Probably a dupe of #1962

knnniggett, 2018-01-14:
We've made a few changes and moved our dockerfiles to a new home:
https://github.com/ZoneMinder/zmdockerfiles

Try that one.

closed 2018-01-14 by knnniggett