ZoneMinder/zoneminder #1990: Errors in logs running ZM version 1.30.4

user111222333, 2017-09-29:
Hello ZoneMinder development team! :)

Thank you for an awesome software you are developing. I'm a big fan and I appreciate it your time and afford. 

I would like to reach out to you to address a couple of possible bugs.

First, this is the server info:

OS: 
Linux nvr01 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u5 (2017-09-19) x86_64 GNU/Linux

WEB_Server: 
Server version: Apache/2.4.25 (Debian)
Server built:   2017-09-19T18:58:57

DB:
mariadb  Ver 15.1 Distrib 10.1.26-MariaDB, for debian-linux-gnu (x86_64) using readline 5.2

PHP:
PHP 7.0.19-1 (cli) (built: May 11 2017 14:04:47) ( NTS )
Copyright (c) 1997-2017 The PHP Group
Zend Engine v3.0.0, Copyright (c) 1998-2017 Zend Technologies
    with Zend OPcache v7.0.19-1, Copyright (c) 1999-2017, by Zend Technologies

ZM verion: 
zmaudit.pl -v
1.30.4

Build guide used: 
https://wiki.zoneminder.com/Debian_9_64-bit_with_Zoneminder_1.30.4_the_Easy_Way

Key points here - this guide was used because it is the only reference to a sucsessful install on new Debian 9 (Stretch) + MariaDB + PHP7

The guide on: 
http://zoneminder.readthedocs.io/en/stable/installationguide/debian.html#easy-way-debian-jessie
...does not mention if ZoneMinder can be installed on a newer OS from Debian project, and use of Debian Stretch is a critical requirement for me. 

Issue: 
Logs are flooded with error messages. ZoneMinder appears to work, however there is inconsistent behavior when 'events' are deleted. The Picture frame still displayed, nothing is happening... and after a long delay (~20 secons) next event is pulled in to the frame. 

There is endless error messages in the log regarding '/tmp/zm'. This directory is present on file system, with the following details:

drwxr-xr-x  2 www-data www-data 4.0K Sep 27 22:22 zm

...So this directory is there and owned by www-data user. Why there is an error in the logs - I do not know ;(

Below is error messages copied from /var/log/syslog directory, the same messages are reflected in GUI served by the www server under 'Log' item. 

Please let me know if you have any questions or/and need more technical info. I would love to contribute and be a guinea pig to help to iron out this error messages. I love your project and the work that you do :)

Sep 27 21:13:06 nvr01 zmc_m2[19997]: ERR [Can't insert log entry: sql(insert into Logs ( TimeKey, Component, ServerId, Pid, Level, Code, Message, File, Line ) values ( 1506571986.528016, 'zmc_m2', 0, 19997, -2, 'ERR', 'Unable to get response, disconnecting', 'zm_remote_camera_http.cpp', 1141 )) error(Lost connection to MySQL server during query)]
Sep 27 21:13:06 nvr01 zmc_m2[19997]: ERR [Unable to get response, disconnecting]
Sep 27 21:13:06 nvr01 zmdc[18952]: ERR ['zmc -m 2' exited abnormally, exit status 255]
Sep 27 21:13:08 nvr01 zmc_m2[20134]: ERR [Error while decoding frame 0]
Sep 27 21:13:08 nvr01 zmc_m2[20134]: ERR [256: 14 85 3c 47 f7 03 0e e6 2f 9c 5a ba d1 d7 6a 22 d3 b6 5e f4 67 89 ab 2c 13 24 07 e9 ff fc 29 f5 47 78 85 cb a8 c1 aa ae ef b8 d8 b2 a9 76 d1 47 60 4a 14 f6 b8 cf 62 90 62 2c 49 64 3f 27 61 07 49 d2 36 31 e8 54 14 fc ec eb 3c 1d b4 ec 97 df be be a0 63 63 24 d9 74 0a 13 f4 0d da 23 7f b8 2c fc c0 48 e8 84 0c 74 29 fc 9d 5a 5c b8 60 b3 82 29 8d b9 80 67 fc 49 13 8c b6 cf 23 e2 35 e0 58 23 f5 b7 2e 0b e1 1f 38 22 6b 10 ef 2d e4 bd 28 84 f7 07 4c 28 8d 91 a6 2b 6a a5 8e de 6d 5e c2 93 e1 4f 1e dc d9 89 53 ec 69 2b f0 0e 5b 0b a0 8d 98 d2 02 01 1b 07 7b 5a 8a a4 6f 9d d2 01 ed a2 3b 5b 48 5b 8d 0a 91 f4 e8 53 fe 3d cb 3f 26 58 dc 72 bb aa 5b ad 69 bf c9 88 e4 f5 a1 b3 0e 02 7c fd e6 93 7d b6 c7 58 89 a4 1a 6d be b6 bc 4d a7 82 9f 38 a6 7d bc 8d b0 7e 72 96 f0 8f]
Sep 27 21:13:08 nvr01 zmc_m2[20134]: ERR [Error while decoding frame 0]

Sep 27 21:13:19 nvr01 web_php[19297]: ERR [Socket /var/run/zm/zms-674568s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
Sep 27 21:13:19 nvr01 web_php[19297]: ERR [getStreamCmdResponse stream error: Socket /var/run/zm/zms-674568s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors()]

Sep 27 21:13:31 nvr01 web_php[19297]: ERR [Socket /var/run/zm/zms-674568s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
Sep 27 21:13:32 nvr01 web_php[19297]: ERR [getStreamCmdResponse stream error: Socket /var/run/zm/zms-674568s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors()]

Sep 27 21:14:51 nvr01 zms[20212]: ERR [Can't stat '/tmp/zm': No such file or directory]
Sep 27 21:14:51 nvr01 zms[20212]: ERR [Unable to validate swap image path, disabling buffered playback]
Sep 27 21:15:43 nvr01 zms[20215]: ERR [Can't stat '/tmp/zm': No such file or directory]
Sep 27 21:15:43 nvr01 zms[20215]: ERR [Unable to validate swap image path, disabling buffered playback]
Sep 27 21:15:43 nvr01 zms[20215]: ERR [Terminating, last frame sent time 1506572143.057616 secs more than maximum of 10.000000]
Sep 27 21:15:43 nvr01 zms[20220]: ERR [Can't stat '/tmp/zm': No such file or directory]
Sep 27 21:15:43 nvr01 zms[20220]: ERR [Unable to validate swap image path, disabling buffered playback]
[... 12 lines cut ...]

Sep 27 22:56:15 nvr01 zmdc[901]: ERR ['zmc -m 2' exited abnormally, exit status 255]
Sep 27 22:56:15 nvr01 zmdc[901]: ERR ['zmc -m 2' exited abnormally, exit status 255]
Sep 27 22:56:29 nvr01 zms[1276]: ERR [Can't stat '/tmp/zm': No such file or directory]
Sep 27 22:56:29 nvr01 zms[1276]: ERR [Unable to validate swap image path, disabling buffered playback]
Sep 27 22:56:30 nvr01 zms[1276]: ERR [Terminating, last frame sent time 1506578189.861496 secs more than maximum of 10.000000]
Sep 27 22:56:30 nvr01 zmdc[901]: ERR ['zmc -m 2' exited abnormally, exit status 255]
Sep 27 22:56:34 nvr01 web_php[754]: ERR [Socket /var/run/zm/zms-997890s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
Sep 27 22:56:34 nvr01 web_php[754]: ERR [getStreamCmdResponse stream error: Socket /var/run/zm/zms-997890s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exi [... cut ...] during query)]
Sep 28 16:56:05 nvr01 zms[8021]: ERR [Can't open /usr/share/zoneminder/www/events/2/17/09/28/16/23/15/00052-capture.jpg: No such file or directory]
Sep 28 16:56:17 nvr01 zms[8036]: ERR [Can't open /usr/share/zoneminder/www/events/2/17/09/28/16/41/51/00053-capture.jpg: No such file or directory]
Sep 28 16:56:31 nvr01 zms[8052]: ERR [Can't stat '/tmp/zm': No such file or directory]
Sep 28 16:56:31 nvr01 zms[8052]: ERR [Unable to validate swap image path, disabling buffered playback]

mullinmatthew, 2017-11-28:
I have almost identical errors, and I followed the same instructions to get Zoneminder running on this set up:
Debian 9.2 QEMU/KVM guest for Zoneminder
A separate Debian 9.2 QEMU/KVM guest using MariaDB to provide SQL to Zoneminder
Debian 9.2 host x86_64

Guests are using a bridged virtio driver. Everything works. The issue for me is the log is rather crowded with errors, I don't know enough to determine if any of these are important, and I wont have much of a heads up if a new problem develops. If anyone is interested, I'll provide logs and detailed configuration info.

connortechnology, 2017-11-28:
So there are a lot of problems here.

The most important one is that the database is going away.  Why?  Maybe look into mysql logs.

second problem is it xcan't write to where it wants to store the videos.  

These guides were written to be followed to the letter.  This is a seriously botched install.

knnniggett, 2017-12-30:
Closing this out as it has gone stale. For support, please post in user forum.

closed 2017-12-30 by knnniggett

user111222333, 2017-12-31:
> -------- Original Message --------
> Local Time: December 30, 2017 6:05 AM
> UTC Time: December 30, 2017 2:05 PM

But no one ever replied to my posted message... ;(
DM

connortechnology, 2018-01-01:
I did.  You ignored it.

pictsidhe, 2018-03-02:
I have the same issue, I followed the same guide, to the letter. It was nearly perfect, I had to mess with the schmall memory not mentioned in the guide to get zoneminder running. 
If someone could point me to a guide to fix this, there are some clueless people out here who would appreciate it.

knnniggett, 2018-03-02:
This thread is being locked for further discussion since it does not describe a bug in the software. Rather, the data presented here indicates multiple known configuration problems. 

When in doubt, always post in the user forum. We will be glad to help you there. 

If during a user-forum discussion data is presented which really does indicate a bug, only then is it acceptable to track the issue here. This is being done to separate real bugs from non-bugs. If let anyone post anything into either forum, then there would be no point in having two different forums.