ZoneMinder/zoneminder #1992: Offline camera, 95% CPU

nitrag, 2017-10-02:
ZM v1.30.4

I have a camera that is wifi challenged and is frequently offline. I have it set to MOCORD/enabled. I was wondering why in `top` one of my `zmc` processes was running 90%+ constantly. I disabled cameras one by one. Sure enough it was the camera that was offline. I re-enabled/disabled several times and sure enough 90%+ each time.

This is a problem as I'm monitoring many, unreliable cameras - don't ask. The main issue though is that ZM is spinning out of control when there is no camera feed though. This is a waste of system resources (and my power bill).

SteveGilvarry, 2017-10-02:
Cannot remember if there is something to stop this already, but you are right it should try a few times and then slow down on the connection attempts. Never see it myself as I run wired cameras that seem pretty reliable. Someone/I will have to trawl the code to see where the connect loop is and if it would be easy to put a configurable wait timer into it.

knnniggett, 2017-10-04:
I am not saying this is not an issue that should be fixed eventually, but the current workaround is to place a value into the max fps and alarm max fps field that is 1 fps *higher* than the stream coming from the camera.

nitrag, 2017-10-04:
Interesting, are you saying that alarm fps will cut off the processing?

knnniggett, 2017-10-04:
Yes, exactly. See the help text for each of these fields.

knight-of-ni, 2018-08-05:
This was closed by accident. Reopening.

connortechnology, 2018-08-05:
Please note that I believe I fixed this in current master.  New behaviour on disconnect is to wait 2 seconds and then try to reconnect.  So it won't just loop infinitely.  I think.

knight-of-ni, 2018-08-05:
Very nice. 
@Dinsmoor @nitrag 
It would be helpful if either or both of you confirm this by testing the latest master branch, soon to be 1.32.0 release.

Dinsmoor, 2018-08-05:
@knight-of-ni 

I would love to, once it's marked stable.

I love the work you do, and God bless.

Dropped you a few bucks for coffee.

@connortechnology Thanks for your work.

knight-of-ni, 2018-10-04:
Now that 1.32 has been released, this should be pretty easy to test. Any takers?

knight-of-ni, 2018-10-06:
I'm going to close out since this appears to be resolved. Please create a new issue if this is not the case.

closed 2018-10-06 by knight-of-ni