ZoneMinder/zoneminder #1996: ERR	Socket /var/lib/zoneminder/sock/zms-743397s.sock does not exist.  This file is created by zms,

Yateenp88, 2017-10-08:
ERR	Socket /var/lib/zoneminder/sock/zms-743397s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit.

Please help to ride our the problem .Using centos 6.3

Yateenp88, 2017-10-08:
i have change path to /zm/cgi-bin/nph-zms . Still not working

knnniggett, 2017-10-09:
This is not a bug. For support, create a thread in the user forum.

closed 2017-10-09 by knnniggett