ZoneMinder/zoneminder #1999: Zoneminder API php 7.1

gfro84, 2017-10-22:
Last night I updated my ubuntu 17.04 (running zm 1.30.4 without issue) to 17.10. ZM API stopped working in both zmNINJA and at /host/zm/api/host/getVersion.json. After tinkering for a while and losing track of what i had changed, I decided to do a fresh install of 17.10. Same problem, zm web interface works but no API. 

I have followed the instructions from http://zoneminder.readthedocs.io/en/stable/installationguide/ubuntu.html verbatim (save from changing php timezone from '/etc/php/7.0/apache2/php.ini' to '/etc/php/7.1/apache2/php.ini'. I get the same problem, no result in firefox at a http 500 in chrome. 

Apache error log
'[Sun Oct 22 16:24:16.128096 2017] [mpm_prefork:notice] [pid 10008] AH00163: Apache/2.4.27 (Ubuntu) configured -- resuming normal operations
[Sun Oct 22 16:24:16.128113 2017] [core:notice] [pid 10008] AH00094: Command line: '/usr/sbin/apache2'
[Sun Oct 22 16:26:58.043582 2017] [php7:error] [pid 26527] [client 192.168.1.190:47860] PHP Fatal error:  Uncaught Error: Call to undefined function apc_fetch() in /usr/share/php/Cake/Ca$
[Sun Oct 22 16:27:04.197425 2017] [php7:error] [pid 26528] [client 192.168.1.190:47862] PHP Fatal error:  Uncaught Error: Call to undefined function apc_fetch() in /usr/share/php/Cake/Ca$
[Sun Oct 22 16:27:05.305794 2017] [php7:error] [pid 26514] [client 192.168.1.190:47864] PHP Fatal error:  Uncaught Error: Call to undefined function apc_fetch() in /usr/share/php/Cake/Ca$
[Sun Oct 22 16:27:10.742776 2017] [php7:error] [pid 26511] [client 192.168.1.190:47872] PHP Fatal error:  Uncaught Error: Call to undefined function apc_fetch() in /usr/share/php/Cake/Ca$
[Sun Oct 22 16:32:27.037091 2017] [php7:error] [pid 26511] [client 192.168.1.190:47922] PHP Fatal error:  Uncaught Error: Call to undefined function apc_fetch() in /usr/share/php/Cake/Ca$
[Sun Oct 22 16:32:30.803545 2017] [php7:error] [pid 26513] [client 192.168.1.190:47924] PHP Fatal error:  Uncaught Error: Call to undefined function apc_fetch() in /usr/share/php/Cake/Ca$
[Sun Oct 22 16:32:43.866732 2017] [php7:error] [pid 27093] [client 192.168.1.190:47926] PHP Fatal error:  Uncaught Error: Call to undefined function apc_fetch() in /usr/share/php/Cake/Ca$'

Given that I have followed the instruction to the letter and I am using a pretty mainstream distribution, I doubt I am the only one to encounter this problem -  but I cant find anything to help.

Grateful for any assistance or guidance.

George

SteveGilvarry, 2017-10-22:
Debian packaging left a major API component out.
https://forums.zoneminder.com/viewtopic.php?p=102944#p102944

closed 2017-10-22 by SteveGilvarry

gfro84, 2017-10-22:
Thanks Steve. Much appreciated and apologies for using your time answering the already answered. 

George