ZoneMinder/zoneminder #2011: CentOS/Red Hat readme needs to mention firewalld

poflynn, 2017-11-26:
By default CentOS has firewalld turned on and blocking ports 80 & 443. I disabled it to get ZM to work. I was going to submit a PR to the RedHat README adding this tot he instructions and am still happy to if you like but I thought ppl might not like the idea of the firewall being disabled. If this is ok LMK and I will submit the PR. I fixed it using the 2 cmds below, for anyone hitting this issue.

`sudo systemctl stop firewalld
sudo systemctl disable firewalld`

cybersteel8, 2017-11-28:
I recently set up ZoneMinder on my Fedora Server and yes, to access the web interface you need to allow httpd through your firewall. You don't necessarily have to disable it completely, just allow the port httpd is running through using firewall-ctl. On my system, the commands are as follows:

`sudo firewall-ctl zone FedoraServer add port 80/tcp
sudo firewall-ctl zone FedoraServer add port 443/tcp`

This is specific to Fedora, and in particular I am using the default firewall zone "FedoraServer" and customizing it. This name changes on a per-system basis.

Magicmann, 2017-12-13:
Going on what cybersteel8 has said, the best way to do this for Centos 7 would be to open the ports, not disable the firewall.

LukeHandle, 2017-12-15:
I suggest we remove the `--zone` option and let it choose the default. If they are doing anything funky, then they should know to check their active zone / `--list-all-zones` and work out what's going on.

Also, `--add-service=http` instead of specific port - functionally the same ofc. but seems sad to not use it (and easier to read) 😄

mateothegreat, 2018-03-24:
To prevent ONVIF discovery from failing, you're going to want to `firewall-cmd --permanent --add-port=3702/udp` btw.

knnniggett, 2018-04-19:
fixed in commit https://github.com/ZoneMinder/zoneminder/commit/411a83133805b9bc89f1f1b0e05b869bbc1188e3

closed 2018-04-19 by knnniggett