ZoneMinder/zoneminder #2029: OPTIONS_RTSPDESCRIBE  not working

jsayres82, 2018-01-08:
I have the OPTIONS_RTSPDESCRIBE option enabled and the setup URL sent seems to be the "Described" URL appended to the URL that was set up in the monitor.  Actually, even when unchecked it appends the "Described" URL.

Below is the handshake from wireshark:

OPTIONS rtsp://192.168.0.10:554/live/av0 RTSP/1.0
User-Agent: ZoneMinder/1.30.4
CSeq: 1

RTSP/1.0 200 OK
CSeq: 1
Date: Tue, 02 Jan 2018 17:21:56 GMT
Public: OPTIONS,DESCRIBE,SETUP,PLAY,PAUSE,TEARDOWN

DESCRIBE rtsp://192.168.0.10:554/live/av0 RTSP/1.0
User-Agent: ZoneMinder/1.30.4
CSeq: 2

RTSP/1.0 401 Unauthorized
CSeq: 2
WWW-Authenticate: Basic realm="streaming server realm"

DESCRIBE rtsp://192.168.0.10:554/live/av0 RTSP/1.0
Authorization: Basic YWRtaW46YWRtaW4=
User-Agent: ZoneMinder/1.30.4
CSeq: 3

RTSP/1.0 200 OK
CSeq: 3
Date: Tue, 02 Jan 2018 17:21:57 GMT
Content-Type: application/sdp
Content-Language: en
x-Accept-Dynamic-Rate: 1
Content-Length: 381

v=0
o=- 86575122 782967157  IN IP4 192.168.0.10
s=live
c=IN IP4 0.0.0.0
t=0 0
a=control:*
m=audio 0 RTP/AVP 8
a=control:rtsp://192.168.0.10:554/audio0
m=video 0 RTP/AVP 96
b=AS:2058
a=rtpmap:96 H264/90000
a=range:npt=0-
a=fmtp:96 profile-level-id=4d001f;packetization-mode=1;sprop-parameter-sets=Z00AH5WoFAFuQA==,aO48gA==
a=control:rtsp://192.168.0.10:554/video0

SETUP rtsp://192.168.0.10:554/live/av0/rtsp://192.168.0.10:554/video0 RTSP/1.0
Transport: RTP/AVP/TCP;unicast
Authorization: Basic YWRtaW46YWRtaW4=
User-Agent: ZoneMinder/1.30.4
CSeq: 4

RTSP/1.0 415 Unsupported Media Type
CSeq: 4

OPTIONS rtsp://192.168.0.10:554/live/av0 RTSP/1.0
User-Agent: ZoneMinder/1.30.4
CSeq: 1

RTSP/1.0 200 OK
CSeq: 1
Date: Tue, 02 Jan 2018 17:21:56 GMT
Public: OPTIONS,DESCRIBE,SETUP,PLAY,PAUSE,TEARDOWN

DESCRIBE rtsp://192.168.0.10:554/live/av0 RTSP/1.0
User-Agent: ZoneMinder/1.30.4
CSeq: 2

RTSP/1.0 401 Unauthorized
CSeq: 2
WWW-Authenticate: Basic realm="streaming server realm"

DESCRIBE rtsp://192.168.0.10:554/live/av0 RTSP/1.0
Authorization: Basic YWRtaW46YWRtaW4=
User-Agent: ZoneMinder/1.30.4
CSeq: 3

RTSP/1.0 200 OK
CSeq: 3
Date: Tue, 02 Jan 2018 17:21:57 GMT
Content-Type: application/sdp
Content-Language: en
x-Accept-Dynamic-Rate: 1
Content-Length: 381

v=0
o=- 86575122 782967157  IN IP4 192.168.0.10
s=live
c=IN IP4 0.0.0.0
t=0 0
a=control:*
m=audio 0 RTP/AVP 8
a=control:rtsp://192.168.0.10:554/audio0
m=video 0 RTP/AVP 96
b=AS:2058
a=rtpmap:96 H264/90000
a=range:npt=0-
a=fmtp:96 profile-level-id=4d001f;packetization-mode=1;sprop-parameter-sets=Z00AH5WoFAFuQA==,aO48gA==
a=control:rtsp://192.168.0.10:554/video0

`SETUP rtsp://192.168.0.10:554/live/av0/rtsp://192.168.0.10:554/video0 RTSP/1.0`
Transport: RTP/AVP/TCP;unicast
Authorization: Basic YWRtaW46YWRtaW4=
User-Agent: ZoneMinder/1.30.4
CSeq: 4

RTSP/1.0 415 Unsupported Media Type
CSeq: 4

Below is my monitor's setup.  
[image]

jsayres82, 2018-01-08:
The current Absolute/Relative URL check on line 452 of zm_RTSP.cpp only checks if the control is identical. After adding a find of "RTSP://" then setting the trackUrl = controlURL I was able to connect. 

Now I am seeing that after setup of the video media stream, it immediately sends a PLAY.  The audio stream is never setup.

knnniggett, 2018-01-08:
Yes, there are many things not quite right with the Remote -> RTSP method.

As stated in our documentation, the Remote -> RTSP method is no longer being maintained:
https://zoneminder.readthedocs.io/en/stable/userguide/definemonitor.html#source-tab

Maintaining this method takes a lot of time and merely re-creates what ffmpeg already does better. We simply don't have anyone available to spend time on this. It might even go away at some point.

Feel free to submit your own pull request if you are willing.

Note that this method never has and never will support audio. Audio is currently supported in the master branch, using the ffmpeg method.

closed 2018-01-08 by knnniggett