ZoneMinder/zoneminder #2033: Return event based conditions api doesn't work

goruck, 2018-01-14:
I'm trying to use the **Return event count based on times and conditions** api but can't seem to get it to work, specifically the **conditions** aspect of it. Based on my testing it looks like the api ignores the condition and only returns json based on the time. 

Here's an example of trying to get the number of events per monitor that were recorded in the last day where there were at least 10 frames that were alarms. I verified through the web browser interface that there were far fewer than the reported number of 146 which is essentially all events over that time period. 

Note the [api documentation](http://zoneminder.readthedocs.io/en/stable/api.html#return-event-count-based-on-times-and-conditions) for this curl command looks incorect and throws an error, the command below is my correction.

```bash
$ curl -b cookies.txt "http://192.168.1.4/zm/api/events/consoleEvents/1%20day/AlarmFrames%20>=:%2010.json"
{
    "results": {
        "1": "146",
        "2": "146",
        "3": "146",
        "4": "146",
        "5": "146",
        "6": "146"
    }
```

Here's an example of just trying to get the total number of events per monitor over the same time period of a day. The values returned look correct based on what I see in the browser UI. 

```bash
$ curl -b cookies.txt "http://192.168.1.4/zm/api/events/consoleEvents/1%20day.json"
{
    "results": {
        "1": "146",
        "2": "146",
        "3": "146",
        "4": "146",
        "5": "146",
        "6": "146"
    }
```

I'm happy to help debug this, just let me know what you need. 

**ZoneMinder Version (`zmaudit.pl -v`):**
```bash
$sudo zmaudit.pl -v
1.29.0
```

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**
No

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**
```bash
$cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 8 (jessie)"
NAME="Debian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=debian
HOME_URL="http://www.debian.org/"
SUPPORT_URL="http://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

**Relevant log lines:**
I don't see any errors raised in the Zoneminder system log when I execute the above curl commands.

pliablepixels, 2018-01-18:
I'm not 100% sure if you need to escape the ">"symbol in curl. Can you try the same thing inside your browser address bar?

In my case, the output seems correct

```
/zm/api/events/consoleEvents/1%20day/AlarmFrames%20%3E=:%2010.json
```
produces 
```
{"results":{"1":"7"}}
```

While:
```
api/events/consoleEvents/1%20day.json
```

produces

```
{"results":{"1":"11","5":"141","7":"141"}}
```

goruck, 2018-01-19:
Thanks for the reply. See answers to your questions below. 

**"I'm not 100% sure if you need to escape the ">"symbol in curl."**

Seems not to matter, I got the same results as above escaping ">" and not escaping it,  both in the browser and from curl. 

**"Can you try the same thing inside your browser address bar?"**

I got the same result in the browser as from curl.

closed 2018-06-22 by stale[bot]