ZoneMinder/zoneminder #2035: Video export regex in export_funtcions.php does not match all video generation options

haake, 2018-01-17:
**ZoneMinder Version (`zmaudit.pl -v`):** 1.30.4

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):** No

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):** CentOS Linux release 7.4.1708 (Core) 

The default distribution offers to generate videos in the following formats:
      mpg,  mpeg,  wmv,  asf,  avi,  mov,  3gp, swf
as specified in the FFMPEG_FORMATS options and the selection control is build dynamically from the db.

The regular expression which finds video files for export  is hard coded on line 837 in export_functions.php:

`if ( preg_match( "/\.(?:mpg|mpeg|avi|asf|3gp)$/", $file ) )`

This expression will not match .mov or .swf files should the user generate them. This results in an empty zmFileList.txt and the creation of an empty zmExport.tar file if tar output is selected or a failure and no export file if zip output is selected.

A quick workaround is to add swf and mov to the regular expression, but a better solution would be to build the regex from the FFMPEG_FORMATS option.

N3X15, 2018-01-17:
Yeah, export also doesn't work properly with MP4, this may be related.  Been trying to export a video of a tree falling on my car since November.

haake, 2018-01-17:
If you are able to generate .mp4 video, then this is likely a problem for your export. Try adding the mp4 extension to the regex in export_functions.php and see if that solves your problem.

PR #2036 (2018-01-18): Add a few more extensions to the list of files to include in export
Fixes #2035
closed 2018-01-18 by knnniggett