ZoneMinder/zoneminder #2044: SQL Error when adding new control

kimmeld, 2018-02-24:
**ZoneMinder Version (`zmaudit.pl -v`):**

1.31.1

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**

Using zmrepo-testing, updated this morning:

```
# yum info zoneminder
...
Installed Packages
Name        : zoneminder
Arch        : x86_64
Version     : 1.31.1
Release     : 1.337.20180224gitgab11e9ec.el7.centos
Size        : 34 M
Repo        : installed
From repo   : zmrepo-testing
```

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**

CentOS Linux release 7.4.1708 (Core)

**Relevant log lines:**
This is shown in the window when clicking Save on a new control method:
```
SQL-ERR 'SQLSTATE[23000]: Integrity constraint violation: 1048 Column 'CanMove' cannot be null', statement was 'insert into Controls set CanMove = NULL, CanMoveDiag = NULL, CanMoveMap = NULL, CanMoveAbs = NULL, CanMoveRel = NULL, CanMoveCon = NULL, CanPan = NULL, MinPanRange = NULL, MaxPanRange = NULL, MinPanStep = NULL, MaxPanStep = NULL, HasPanSpeed = NULL, MinPanSpeed = NULL, MaxPanSpeed = NULL, HasTurboPan = NULL, TurboPanSpeed = NULL, CanTilt = NULL, MinTiltRange = NULL, MaxTiltRange = NULL, MinTiltStep = NULL, MaxTiltStep = NULL, HasTiltSpeed = NULL, MinTiltSpeed = NULL, MaxTiltSpeed = NULL, HasTurboTilt = NULL, TurboTiltSpeed = NULL, CanZoom = NULL, CanZoomAbs = NULL, CanZoomRel = NULL, CanZoomCon = NULL, MinZoomRange = NULL, MaxZoomRange = NULL, MinZoomStep = NULL, MaxZoomStep = NULL, HasZoomSpeed = NULL, MinZoomSpeed = NULL, MaxZoomSpeed = NULL, CanFocus = NULL, CanAutoFocus = NULL, CanFocusAbs = NULL, CanFocusRel = NULL, CanFocusCon = NULL, MinFocusRange = NULL, MaxFocusRange = NULL, MinFocusStep = NULL, MaxFocusStep = NULL, HasFocusSpeed = NULL, MinFocusSpeed = NULL, MaxFocusSpeed = NULL, CanIris = NULL, CanAutoIris = NULL, CanIrisAbs = NULL, CanIrisRel = NULL, CanIrisCon = NULL, MinIrisRange = NULL, MaxIrisRange = NULL, MinIrisStep = NULL, MaxIrisStep = NULL, HasIrisSpeed = NULL, MinIrisSpeed = NULL, MaxIrisSpeed = NULL, CanGain = NULL, CanAutoGain = NULL, CanGainAbs = NULL, CanGainRel = NULL, CanGainCon = NULL, MinGainRange = NULL, MaxGainRange = NULL, MinGainStep = NULL, MaxGainStep = NULL, HasGainSpeed = NULL, MinGainSpeed = NULL, MaxGainSpeed = NULL, CanWhite = NULL, CanAutoWhite = NULL, CanWhiteAbs = NULL, CanWhiteRel = NULL, CanWhiteCon = NULL, MinWhiteRange = NULL, MaxWhiteRange = NULL, MinWhiteStep = NULL, MaxWhiteStep = NULL, HasWhiteSpeed = NULL, MinWhiteSpeed = NULL, MaxWhiteSpeed = NULL, HasPresets = NULL, NumPresets = NULL, HasHomePreset = NULL, CanSetPresets = NULL, Name = 'Amcrest', Type = 'Local', Protocol = 'amcrest781''
```

**Replication Steps**
Background:  I am trying to add a new remote control method for an Amcrest camera.  I'm using the script found in [this thread on the forums](https://forums.zoneminder.com/viewtopic.php?f=9&t=25606).

1.  Go to Monitor Settings > Control tab, click Edit beside the Control Type, click Add New Control.
2.  Fill in Protocol = amcrest781
3.  Click Save.

Expected result:  The control saves successfully.

Actual result:  Any boolean fields which have not been selected try to get set to NULL in the database.  Any text fields which have not been filled in try to get set to NULL in the database.  Varying the fields selected varies the error message as expected, however unless every field is filled in or checked off the record won't save [NOTE: I haven't actually tried this].

I'm guessing that there's logic somewhere that should default the boolean fields to false and numeric fields to 0 if they don't come back from the browser when clicking save, but I haven't been able to unravel this bit of code yet.

knnniggett, 2018-02-25:
log into mysql from the command line, then show the results of `select @@sql_mode;`

kimmeld, 2018-02-25:
Select @@sql_mode shows NO_ENGINE_SUBSTITUTION

knnniggett, 2018-02-25:
This is a regression. 1a4333411381d7622c8e82845c110fa78e88c345 is the offending commit. You can fix this today by just editing your php file back to the way it was prior to that commit. I'm waiting to hear back from @connortechnology to determine whether we will set this back permanently or do something else.

kimmeld, 2018-02-26:
I reverted that file and was able to add the new control method.  Thanks!

I'll leave this open since the underlying issue still needs to be worked on.

knnniggett, 2018-02-28:
Closing, since #2046 fixes this.

connortechnology, 2018-02-28:
That PR didn't ger merged.  I deleted it because it had some other unrelated cruft.  I have yet to regenerate it.

knnniggett, 2018-02-28:
oh sorry, I must not have looked at it closely.

knnniggett, 2018-03-03:
Closing this for real this time since the work by @connortechnology was merged. This change should show up in zmrepo in the next 24-48 hours. Please test. Feel free to open a new issue if anything new crops up.

closed 2018-03-03 by knnniggett

knnniggett, 2018-03-04:
Argh.... so it might be a couple more days before you see this in zmrepo.
@connortechnology Internet is down which means zmrepo is down to. We do have mirror servers so end users should not see any yum errors, but no packages can be deployed until zmrepo comes back.

@connortechnology was last heard cursing the roofers at his house that may have been a little too aggressive at their work.

knnniggett, 2018-03-11:
@kimmeld I'm see aborted connection errors in my sql logs and the enable checkbox no longer saves when I run the latest development package in zmrepo. I actually saw it first in a new feature I was developing and switched back to the master branch to try to isolate where the problem was coming from. 
I think these are related to the recent commit, which I will investigate further.

Let me know if you see the same problem on your end. All you have to do is open a monitor and toggle the enable checkbox, then hit save. Let me know if you can turn it back on.

kimmeld, 2018-03-12:
@knnniggett I think I'm seeing the same behaviour.  When I click on the Source and try to enable a disabled camera it doesn't work.  When I click on the Function I can enable and disable the camera like normal.

I didn't see any errors in the logs, but I may not be looking in the right spot.

Using 1.31.1-1.370.20180307gitg0d9f5b73.el7.centos from zmrepo-testing, which includes this commit.

knnniggett, 2018-03-12:
I pushed a couple of commits through that fixed the php issues I was seeing. New packages should appear in a day or so.

After further testing, the sql issue appears isolated to my fedora 27 vm so that is probably not related.