ZoneMinder/zoneminder #2055: Rotating a camera does not rotate detection zones

doublehp, 2018-03-07:
Repro:
- add a new monitor
- view it
- click on '1' in the zones column
- check the zone covers the whole image; it usually is the case by default; cancel the windows; no need to make any change in there, just check
- back in main page, click on the IP of camera to change settings
- in the source tab, orientation line, change for ROTATE LEFT (i did not try rotate right, but I bet it has the same bug)
- save
- go again in the zone editor
- click on the zone
=> if you camera produces a rectangular image (I have never seen a square CCD in my life; I know hey exist, but never met any), then two of the four corners are unreachable, and unmovable with mouse.

There are two workarounds:
- return in camera settings, put the cam in normal orientation, return in zone editor, make the zone very small, set rotation again, go in zone editor, now all 4 corners are inside the image
- use the bottom table to manually alter coordinates (needs experience to guess which point is where; not user friendly at all).

# zmaudit.pl -v
1.30.0

# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux stretch/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

But, since you said ZoneMinder is mature, and unlikely to have undiscovered bugs, the issue is probably a PEBCAK like you always stated in the past. Please close this bug ASAP without fixing it.

Any way, I have bought a HIK Vision NVR, and now moving my cams from ZM to HIK. It's not perfect, but they officially refuse to listen for my bug reports, and do not pretend to have a public BTS; they don't insult me for trying to help them fixing theirproducts.

doublehp, 2018-06-16:
So convenient to auto-close bugs because you don't fancy fix them ...

knnniggett, 2018-06-16:
@doublehp Yeah, that's right. When there are only two of us to willing to work on this project in our spare time, there is an ever growing list of work that doesn't get done. There is a very limited amount of work that is going to be accomplished in this scenario. What we need are people willing to do real work on the project. The last thing we need is some entitled jerk in our forum who never has anything good to say and cries because no one wants to work on his bug report. Congratulations, your attitude and past rude comments have earned yourself removal from this forum.

closed 2018-06-16 by knnniggett