ZoneMinder/zoneminder #2057: Can not set Source type to FFMPEG on new camera

doublehp, 2018-03-08:
Some forum said that in some case, source type REMOTE works better than FFMPEG.

So I have remove a camera, and reinstall it from scratch.

- add new monitor
- ONVIF
- select device in list
- save
- select any substream
- save
- change source type from FFMPEG to REMOTE
- save
=> "Names may only contain alphanumeric characters plus spaces, hyphen and underscore
Host must be set to a valid ip address or hostname, do not include http://"

... I did not enter any of that. Yes in the REMOTE tab the "Remote Host Name" field is set to "http://192.168.246.141/onvif/device_service", but I never wrote this there; it's ZoneMinder which detected and produced this setup; if there is any thing wrong, or uncompatible, ZM has to sort it out by itself. I have also tried to alter this field to "192.168.246.141/onvif/device_service" and "192.168.246.141" . The last option allowed me to save the config, but produced a blue image (while leaving FFMPEG produces a real image).

SteveGilvarry, 2018-03-08:
Core issue is Onvif is automatic configuration for ffmpeg cameras, fix is to not allow users to flip between the two, but some people like the ability to switch between camera methods and do the tweaking that is required. Not sure anyone is actively developing Remote so maybe easier to turn off.
Thanks for raising the issue, workaround for now is don't change Onvif configured cameras

doublehp, 2018-06-16:
...

knnniggett, 2018-06-16:
The Remote -> RTSP has been deprecated. This method is no longer being maintained by us, and ZoneMinder 1.32.0 will be the last version with this source type. Moving forward, please use the ffmpeg source type.

closed 2018-06-16 by knnniggett