ZoneMinder/zoneminder #2059: alarm_cause in SharedData

pliablepixels, 2018-03-10:
As of today, SharedData has a field called 'trigger_cause' that holds a text value when you create an external trigger, or 'ForcedWeb' if you force it from the web. There is no equivalent field for 'alarm cause' when its not triggered (i.e. regular alarm). The notes are directly written to the DB. Would it be useful to either overload 'trigger_cause' with 'alarm_cause' or create a new field? I'm not sure what the latter involves as we are changing the size of SharedData and not sure if other areas depend on this. If its the former, I can do a PR, but I do agree it would be breaking the core idea of 'trigger_'

Motivation: zmeventserver (used by zmninja) relies on shared memory to get alarm details and this would be useful as a push notification

connortechnology, 2018-03-10:
Sounds like a good idea to me.  I don't see any technical reasons not to.

pliablepixels, 2018-03-10:
@connortechnology do you mean adding a new field or overloading trigger_cause? `SharedData` has [comments](https://github.com/ZoneMinder/zoneminder/blob/dd70eb3e5f80edbf87280e5ccaf7680b26b8e021/src/zm_monitor.h#L98) about expecting an exact size. Is adding a new field a good idea?

connortechnology, 2018-03-10:
Adding new fields isn't a bit deal.  It just changes the size of the shared data.  Which isn't a big deal.  It means that you can't mix older binaries with new ones... which you already can't because we don't allow it.  That's why we die when the version doesn't match or # of config items doesn't match what is expected.  

I would add a new field.  The code that manages the notes is a little messy and may get messier when using a simple char field in shared mem.  So be it.

closed 2018-03-13 by pliablepixels