ZoneMinder/zoneminder #2064: Postinstall script of latets update fails

jzielke-nli, 2018-03-16:
After updating zoneminder the postinstall script runs insto an error

**Relevant log lines:**
```
Setting up zoneminder (1.31.1~20180315113240-xenial) ...
/var/lib/dpkg/info/zoneminder.postinst: 50: /etc/zm/zm.conf: tk#S: not found
dpkg: error processing package zoneminder (--configure):
 subprocess installed post-installation script returned error exit status 127
Errors were encountered while processing:
 zoneminder
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

MTres19, 2018-03-17:
Hmmm... seeing as I wrote the updates to the postinst script, I feel a bit responsible.
However, the error makes it look like you have the line "tk#S" somewhere in your zm.conf. Is that the case?

connortechnology, 2018-03-17:
There is an eerie if you don't have mariadb installed that causes the script to die. More testing and improvement needed

MTres19, 2018-03-17:
Ah, got it. :/ I let a bashism sneak into the script. In bash, if a command
in a command substitution fails it won't kill the script even in set -e
mode. Switching from #!/bin/sh to #!/bin/bash would work, though it's
non-ideal and Ubuntu probably wouldn't like it. One way to fix it would be
to use systemctl status mariadb.service >/dev/null 2>&1 ||
MARIADB_STATUS=$? instead. Not sure if there are other places with this
problem though.

MTres19, 2018-03-17:
A (relatively) quick look into this seems to indicate that there are two problems: the error when there's no mariadb and the error that the OP mentioned. My error seems to be from the aforementioned bashism, but looking at the systemctl manpage there's a better way to do this anyway. I'll try to get this fixed and submit a PR.

MTres19, 2018-03-18:
@connortechnology I tested my changes that are in #2066 and the installation seems to work fine on my end. Sorry for causing this bug.

jzielke-nli, 2018-03-18:
Never mind, we all make mistakes :-) Thanks for fixing it so fast 👍

jzielke-nli, 2018-03-31:
I actually digged into my config file again and noticed, that the post-install script had problems with a `#`character in my passphrase, since the phrase was unquoted.

MTres19, 2018-04-05:
I don't think I made any significant changes (if any changes at all) to the way `mysql` gets called in the postinst script. Do you have unix_socket authentication disabled for the root database user? The script relies on that, because it's not safe to provide passwords to the `mysql` command on the command line. With unix_socket authentication, MySQL/MariaDB allows connections under the root db username as long as the process that is accessing the database is running as the root system user. It might be possible to have the postinst script generate an option file for the mysql command, though that would add unfortunate complexity.

MTres19, 2018-04-05:
Actually line 59 already exposes the user's database credentials on the command line.... Not sure how you want this to be fixed @connortechnology. (It was like this before my PR.)

closed 2018-06-22 by stale[bot]