ZoneMinder/zoneminder #2070: Missing point highlights in zone editing in v1.31.39

xenoxaos, 2018-03-30:
Two separate installs (I was testing the storage areas PPA on a new VM) and one thing I'm seeing is that the indices for editing a zone are no longer highlighted green in v1.31.39. Is there something I'm missing?
[image]
[image]

knnniggett, 2018-04-20:
@connortechnology Do you know if this still the case, following our recent changes?

connortechnology, 2018-04-20:
I'm not able to reproduce it, although the last time I saw this it had to do with the css files using ../../../ to get to the images and when you don't have /zm in the url it changes the path.  So I made the decision that no matter how zm is accessed, either with the /zm or without, /zm would still work (by using an alias in apache config).  So now all the images should work.  However it may be a good idea to do some testing.

knnniggett, 2018-06-15:
@xenoxaos  This is working for me on latest master. Please let us know if you still experience this issue.

knight-of-ni, 2018-07-03:
I'm closing this out despite not receiving feedback from the op. This is working for us.

closed 2018-07-03 by knight-of-ni