ZoneMinder/zoneminder #2071: Unable to enter Monitor - mysql

asquith, 2018-04-01:
Running zm 1.30.4

Unable to create monitor, error in query.  Mysql insert query has empty strings for nullable columns.  Strings are incompatible with data types.  Had to convert them to varchar.

knnniggett, 2018-04-01:
We did make some changes to the latest master branch, which might cause symptoms like that, but you are running the 1.30.4 release. The more common problem with 1.30.4 is an incorrectly set sql_mode. That will cause problems like this. 

I'm closing this out since you've not given us any data to work with. Feel free to produce real data indicating otherwise. You can also post in the user forum to get help troubleshooting your system.

closed 2018-04-01 by knnniggett

asquith, 2018-04-02:
Hi there,

I know you closed this but just in case you wanted more info.  I get the following error when inserting the event id.

SQL-ERR 'SQLSTATE[HY000]: General error: 1366 Incorrect integer value: '' for column 'MaxPixelThreshold' at row 1', statement was 'INSERT INTO Zones SET MonitorId=?, NumCoords = '4', Coords = '17,47 86,45 88,146 16,120', Area = '6128', AlarmRGB = '16711680', Name = 'New', Type = 'Active', Units = 'Percent', CheckMethod = 'Blobs', MinPixelThreshold = '25', MaxPixelThreshold = '', FilterX = '3', FilterY = '3', MinAlarmPixels = '183', MaxAlarmPixels = '4596', MinFilterPixels = '183', MaxFilterPixels = '4596', MinBlobPixels = '122', MaxBlobPixels = '0', MinBlobs = '1', MaxBlobs = '', OverloadFrames = '0''

knnniggett, 2018-04-02:
I can call that sql statement just fine from the command line:
```
MariaDB [zm]> use zm;
Database changed
MariaDB [zm]> INSERT INTO Zones SET MonitorId=1, NumCoords = '4', Coords = '17,47 86,45 88,146 16,120', Area = '6128', AlarmRGB = '16711680', Name = 'New', Type = 'Active', Units = 'Percent', CheckMethod = 'Blobs', MinPixelThreshold = '25', MaxPixelThreshold = '', FilterX = '3', FilterY = '3', MinAlarmPixels = '183', MaxAlarmPixels = '4596', MinFilterPixels = '183', MaxFilterPixels = '4596', MinBlobPixels = '122', MaxBlobPixels = '0', MinBlobs = '1', MaxBlobs = '', OverloadFrames = '0';
Query OK, 1 row affected, 2 warnings (0.02 sec)

MariaDB [zm]>exit
```

The sql_mode parameter is set incorrectly on your machine. Find the ZoneMinder installation instructions for your linux distro. It will tell you how to set the sql_mode, along with other steps necessary for a properly functional system.

asquith, 2018-04-02:
SET GLOBAL sql_mode='';

That did the trick.

knnniggett, 2018-04-02:
You may run into other sql related issues if you don't set it to:
`sql_mode=NO_ENGINE_SUBSTITUTION`

Make sure and follow the ZoneMinder installation instructions for your distro, or you will likely run into other common problems.