ZoneMinder/zoneminder #2072: JavaScript Error With Video Feed

ac3d912, 2018-04-01:
https://github.com/ZoneMinder/zoneminder/blob/06f10e0ed30450dbbc60afb6c32369512c001694/web/skins/classic/views/js/event.js#L972-L977

Throws an exception when imageFeed does not exist, as in when there is a videoFeed instead.  It's a few lines to fix, I can submit a PR or one of the routine maintainers can add this fix to their current dev branch.

```
    if ( canStreamNative ) {
      var streamImg = $('imageFeed');
      if ( streamImg ) {
        streamImg = streamImg.getElement('img');
        if ( !streamImg )
          streamImg = $('imageFeed').getElement('object');
        $(streamImg).addEvent( 'click', function( event ) { handleClick( event ); } );
      }
    }
```

knnniggett, 2018-04-20:
We've recently made a lot of changes to the master branch, but it does not look like the code being referenced here was affected. Please test this again using the lastest master branch. If the problem persists, please submit a pr and we'll get this into the next release. Thanks for your help.

ac3d912, 2018-04-21:
TL;DR: Closing, fixed with some other change.

So it appears in the version I was testing, for some reason the <div ... id="videoobj"> on the event page wasn't correct.  In the current version 1.31.42, the id="videoobj" is there, so it takes the first block of that if/else block... at any rate, looks like it all works now.

closed 2018-04-21 by ac3d912