ZoneMinder/zoneminder #2073: stream.php lock file name issues - affects control commands

pliablepixels, 2018-04-11:
stream.php received control commands from browsers/apps for existing streams. There is a problem with [this](https://github.com/ZoneMinder/zoneminder/blob/master/web/ajax/stream.php#L15) line of code where it looks for a lock file that doesn't exist (the file created is `connkey.lock` not `connkeyw.lock`. This causes problem when control commands for different connkeys are sent - they error out. zmNinja relies on this to ensure there are no network and memory leaks. I'll submit a tiny PR to fix this.

knnniggett, 2018-04-12:
Actually, now I'm confused. Your description talks about a lock file, but the line of code you are attempting to modify creates a .sock file. The lock gets created in control_functions.php. 

Can you clarify?

connortechnology, 2018-04-12:
It is super confusing all this lock stuff.  

Also, @pliablepixels have been figuring this out on my branch, which has a lot of other stuff in it as well, which adds to the confusion.  

This PR is not correct and I think we will need to do a larger port from my branch.

pliablepixels, 2018-04-12:
Yup, I should have close the associated PR. The issue is legit, the fix is not. Isaac fixed it the right way in his branch. I also need to resolve the pause issue.

knnniggett, 2018-04-18:
soooo..... does that mean this issue can be closed?

pliablepixels, 2018-04-18:
yes, thx

closed 2018-04-18 by pliablepixels